onlaj/Piano-LED-Visualizer #91: Nothing is working

eengamer, 2020-03-06:
Hello, i have recently got my pieces and out ecerything togheter the first time my pi died and i got a replacement. Now when i try it the pi gets really hot the program freezes instantly the leds have neverworked on the pi. I use a 5v 10a psu from meanwell i have tried alot do i need to get another replacement or is the sd faulty. Pi gets hot even without sd card. This is getting pretty annoying for me since nothing works. Hope you can help me?

onlaj, 2020-03-07:
Did you try running test programs from here?
https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/

eengamer, 2020-03-08:
I have tried evrything but it wont work my second raspberry pi is also broken it get so hot that it almosr burns you the program freezes immideatly wenn i get an replacement can we get in a call an make together?

vzoltan, 2020-03-08:
Raspbian lite or full desktop?

eengamer, 2020-03-08:
Full  the programm works but when i connect the leds noting works and 2 time my pi broke now it starts get really hot freezes and reboots

onlaj, 2020-03-08:
Are you sure you connected the wires correctly? Try to unplug and check if RPi is getting hot without them.

eengamer, 2020-03-08:
It gets hot without them because it brike when i had everything connected and the lights wouldnt go on or work

Jaxii, 2020-03-09:
I suggest buying a new Pi, perhaps one from a reputable supplier.

eengamer, 2020-03-09:
Is there anyone that could help me as soon as i get my replacement pi??

closed 2020-03-16 by eengamer