onlaj/Piano-LED-Visualizer #93: After a minute the program stop working

Roloo38, 2020-03-12:
Hello, after a minute that I run the code, the screen no longer works, it remains blocked, and the LEDs either do not work or remain blocked on the piano keys.
If you think it will help, i can share a video with you.
[image]
pressed last time

onlaj, 2020-03-12:
What does the console say?

Roloo38, 2020-03-12:
Nothing. I attach a photo. Sometimes it works...sometimes it doesn’t 
[image]

Roloo38, 2020-03-12:
I think I figured it out...i was using a iphone terminal, and when the app shut down, the raspberry freeze..sorry for bothering you

closed 2020-03-12 by Roloo38

vzoltan, 2020-03-12:
either put & at the and, or run it as a service.

Roloo38, 2020-03-12:
Ok, thanks!