onlaj/Piano-LED-Visualizer #94: Sugestion for leds wires and screen

Roloo38, 2020-03-12:
for those who use raspberry 2/3/4 b + and do not want to solder the wires on the pins, under the screen, I found a cheap and good solution.  this pin extension board can be used to connect the screen but also to connect the LEDs to the pins. I will attach some photos and a link.

RON 9.34  24％ Off | Raspberry Pi GPIO Extension Board 1 to 3 40 Pin GPIO Module for Orange Pi PC Raspberry Pi 4B/3B+/3B Raspberry Pi Zero W/1.3
https://a.aliexpress.com/_dVErxlT

* i put some tape on the second row of pins to not touch my screen
![EF00AB94-F38F-4F85-8FC9-D0057432213D](https://user-images.githubusercontent.com/55906704
[image]
/76562632-76e49380-64ae-11ea-8b8e-c49b15ec6185.jpeg)

[image]

[image]

onlaj, 2020-04-02:
I used something like this
[image]
and it worked too, but it wouldn't fit the case.

What version of RPi do you use? Do you have any latency issues?

vzoltan, 2020-04-02:
@Roloo38 hi

could you / would you pls share the source of your led profile?
i cant find a nice thin and clean like yours or @onlaj 's  (only in poland, but that store wont deliver to me :( )

Thank you

Roloo38, 2020-05-26:
@onlaj hi, and sorry because i didn’t responded. I use raspberry pi 3B+, and i don’t have any latency problem, everything works fine :). My gpio adaptor fits perfectly on rpi, and in the case

Roloo38, 2020-05-26:
@vzoltan hi, sorry for late response. I buyed my led profile from a store in my country (Romania), and i don’t think they deliver out of the country. Sorry :(

closed 2020-12-23 by onlaj