onlaj/Piano-LED-Visualizer #95: Script not booting up

eengamer, 2020-03-16:
Hello today the script and led worked fine i also today added samba file sharing to the pi after that i pluged it in and everything worked fine. I then shutted it off and when i came back to play again it was on white screen zo i unplugged it and put it back in again and it wont boot up to the programm

eengamer, 2020-03-16:
Note i also did switch from autologin console to desktop amd then switched back

eengamer, 2020-03-17:
[image]
This is what i says when i plug it in

eengamer, 2020-03-17:
[image]
This is what it says

onlaj, 2020-03-17:
Looks like your settings.xml is damaged, try to clone it again.

eengamer, 2020-03-19:
did it and it now works

closed 2020-03-19 by eengamer