onlaj/Piano-LED-Visualizer #96: midi playback to fast

eengamer, 2020-03-19:
when i transfer mz recorded midi from the py on to mz pc it doesnt plazback ate the same speed as i plazed this can be reallz annoying when you want to make an video is there anz fix for this???

CaptainSword123, 2020-03-22:
no fix found, but same problem:/

onlaj, 2020-03-23:
I fixed midi playback, so maybe I will be able to also fix recording using the same method.
For now if you want to record midi you can buy this http://compasflamenco.com/midi-c-3/midi-usbusb-p-4.html , connect RPi to your PC and use some midi recording software.

AnimusAura, 2020-04-14:
Hello. So we need an USB otg splitter to connect everything together? I mean if I want to use the piano with led visualisation, I need to connect the piano with the RPi (USB).
If I want to record midi with my PC, I need to connect the piano with the PC (USB). 
What if I want to do this things at the same time? 

If I want to make a video like Rousseau made, I need the LED visualisation when I play the piano but need that midi too.
Because with the recorded midi I can do the upper half of the video, that synthesia kind of effects. 

I need some kind of USB otg splitter to use visualisation and record midi (with pc) at the same time? 
(I want not use BT,  because I need a perfect midi without delay, miss or too fast playback) 

Why we need that linked USB-midi-USB thing? It has 2 USB ports, or some kind of 3rd port too? 

Sorry if this questions are stupid, but I want just clearing things. 

Anyways this is a really great project, thank you for your great work  :-)

CaptainSword123, 2020-04-14:
Hey AnimusAuro,
A normal usb splitter won‘t work, because that’s not how usb works. To get information, the computer sends a request and gets an answer. If you connect two computers it is just confused. Fortunately we have midi, and midi is splitable. This have to be done before it’s converted to usb. You could use a midi interface, but these are very expensive. The best solution is the device linked above.

I also enjoy the work and the fast support:)

onlaj, 2020-04-14:
@AnimusAura
> Why we need that linked USB-midi-USB thing? It has 2 USB ports, or some kind of 3rd port too?

If you want to send midi messages through USB at least one of the connected devices must appear as MIDI hardware. It is possible to make RPi emulate MIDI but it would work only for one port and we need second one to connect Piano.

CaptainSword123, 2020-04-14:
If we can emulate the midi signal, couldn‘t we just emulate it on the computer and send it to the raspberry? Or is the delay to long.

onlaj, 2020-04-14:
I think you can't emulate MIDI on most devices. RPi ZERO is an exception because usb is directly connected to CPU. RPI 2/3/4/etc use usb hub therefore it's not possible.

https://gist.github.com/gbaman/50b6cca61dd1c3f88f41

> The Raspberry Pi Zero (and model A and A+) support USB On The Go, given the processor is connected directly to the USB port, unlike on the B, B+ or Pi 2 B, which goes via a USB hub.

CaptainSword123, 2020-04-14:
Oh okay, thank you

AnimusAura, 2020-04-15:
Thank you for the fast and informative answers :-)

I am lucky because my piano can record midi when I play and after that i can export that midi file.

onlaj, 2020-04-26:
@eengamer  @CaptainSword123  @AnimusAura 
I managed to fix that problem.
Some slight drift might still appear if you play very fast, but unless your name is Valentina Lisitsa or Lang Lang, you have nothing to worry about ;)

closed 2020-04-26 by onlaj

CaptainSword123, 2020-04-26:
Oh nice, your work is amazing, thank you very much for all this. How do I update my raspberry?