onlaj/Piano-LED-Visualizer #97: My ledstrip not working

eengamer, 2020-03-26:
Hello onlaj my led strip worked fine till today but today it stopped working ik made sure all connections are good. And i can play the La campanella file on the pi to piano. But mynkeds are not worki
[image]
[image]
[image]n

ng at all

vzoltan, 2020-03-26:
try with the PIN #12  (the green cable)
also modify change it in visualizery.py to 12 from 18
and give it a try again. 

This line:
`self.LED_PIN        = 18      # GPIO pin connected to the pixels (18 uses PWM!).`

eengamer, 2020-03-26:
What do you mean witch pin is 12 and where do i change it

vzoltan, 2020-03-26:
re-wire to pin12 from pin18 on the pi, physically .

eengamer, 2020-03-26:
But where is pin 12 and where can i find the commdan line to change soory im not so good in this

vzoltan, 2020-03-26:
gpio12
[image]

eengamer, 2020-03-26:
How do i accses visualiser.py?

eengamer, 2020-03-26:
I do have telnet enabled

eengamer, 2020-03-26:
[image]
Like this

eengamer, 2020-03-26:
[image]
Still doesnt work

vzoltan, 2020-03-26:
then check somehow your ledstrip if it works or not.

eengamer, 2020-03-26:
The led strip worked fine yesterday and today when i put the piano and things on it complete did nothing

eengamer, 2020-03-26:
It is not the led strip i teste an other one that also didnt work

onlaj, 2020-03-26:
Can it be your power supply? As long as you don't light more than few leds at the same time you should be able to power it directly from RPi, just for testing.
[image]

eengamer, 2020-03-26:
Do i need to run anything for it? Like an script

onlaj, 2020-03-26:
Yes, you can find examples here: https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/
Just change number of leds to something low, like 10, so RPi can power it.

eengamer, 2020-03-27:
I think its the powersupply i will try and get a replacement

eengamer, 2020-03-27:
Hey i think its the psu i didnt do a command just connected everyrhing and ran the visualiser.pythe furst led was blinking really so i know its the psu

eengamer, 2020-04-02:
Hello onlaj i have got my new powersupply today and it still isintnworking what could be the problem i have reinstalled the script bit still no working leds. 1 led on my strip looks like this mayby that could be it but i dont think that bcs the other ledstrip doesnt have that
[image]

onlaj, 2020-04-02:
You said that without power supply it works, can you confirm?

eengamer, 2020-04-02:
I didnt change anything just put the effect on rainbow and connevted it the first led was softly flickering so it should be fine but i cant seem to find the problem

onlaj, 2020-04-02:
Can you connect it like on last image I send, so the whole strip is powered directly by Pi?

eengamer, 2020-04-02:
Thats what i have done and then powered the script and started rainbow effect it was flickering really softly

onlaj, 2020-04-02:
So it doesn't work properly, right? Then it must be either led strip or your Pi, but you said it's the same with two different strips, so it must be RPi. Does it make sense?

eengamer, 2020-04-02:
I think it could be the led strip since the first led looks burnt as shown in the picture

vzoltan, 2020-04-02:
if that bothers, cut it off, and re-solder the wires

eengamer, 2020-04-02:
I dont have a soldering pen but i have to send the led strip back so they can provide a new one

closed 2020-05-04 by eengamer