onlaj/Piano-LED-Visualizer #98: Playback issue

mlewis-uk, 2020-03-27:
When I playback via synthesia I only get some of the notes of a song being played and the corresponding lights - I have spent days checking everything, can you please assist. Using a macbook pro for Synthesia and a Gear4music JDP-1 piano.

onlaj, 2020-03-27:
Are you using bluetooth? Wireless connection may lose some messages.
I recommend this device to connect your PC/Mac with RPi http://compasflamenco.com/midi-c-3/midi-usbusb-p-4.html

mlewis-uk, 2020-03-27:
Hi - thank you for the quick reply - the piano is connected to the PI via the USB port (OTG adapter). The mac/synthesia is via the bluetooth midi.

mlewis-uk, 2020-03-27:
Just as an update, if I play some of the easy songs, like Aura Lee I get the lights but no audio at all.

closed 2020-03-27 by mlewis-uk