onlaj/Piano-LED-Visualizer #99: Script crash

mlewis-uk, 2020-03-27:
I have a 61 key piano - if I follow the guide and set the LEDs to 122 (four occurances in the script) and the offset to 30, 29 and 28 to get the first key over to the left of the LED strip I get the below error:
In key test my first key in printmidimessages is 36 and the last is 96.

Inport set to USB Device:USB Device MIDI 1 20:0
playport set to USB Device:USB Device MIDI 1 20:0
Traceback (most recent call last):
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 1880, in <module>
    keylist_status[(note - 20)*2 - note_offset] = 0
IndexError: list assignment index out of range

mlewis-uk, 2020-03-27:
Ive also noticed LEDs getting stuck on and not always displaying when playing the piano - I have checked in the midi messages and can see the keys are being registered.

This is a great project, amazing bit of work - I'll keep looking to see if I can fix anything myself and uodate if I can. In the mean time if anyone else has these issues and resolved them please advise...

onlaj, 2020-04-02:
Hey, sorry for late reply.
What exactly did you edit at offset part? You should have edit the statement part, not the note_offset variable.
In your case it should look like this, note_offset stay the same as in my code:

```
if(note > 72):
    note_offset = 2
elif(note > 35):
    note_offset = 1
else:
    note_offset = 0
```

I made up the numbers, try others until it fit your keyboard.

mlewis-uk, 2020-04-02:
No problem, thanks for the reply.
I had changed the wrong part, so thank you for the clarification.

The problem I have now is that the starting LED is not in line with my first key, its about 30cms to the right. Is there a way to shift the start position?

Also I am getting the LEDs sticking on still, has anyone else had this?

onlaj, 2020-04-23:
Yes, you can change the offset to lower values.
Something like this:
```
if(note > 72):
    note_offset = -32
elif(note > 35):
    note_offset = -31
else:
    note_offset = -30
```
And again, I made up the numbers, you need to try until they are aligned. Negative offset means that leds will be shifted to the left (assuming that your led strip is going from left to right).

mlewis-uk, 2020-04-23:
Thank you for the reply - If I set the figures to minus it went the other way, further right. I set them to 33, 32 and 31 and now the first LED lights up.

I have a strange one now though. In Synthesia if I play the song from the song selection screen it lights the correct notes, also if I play the song under 'watch and listen'. If I try to practice, whatever hand I am practising is not lighting up? Any suggestions, I am so close to getting this now thanks to your help.

mlewis-uk, 2020-04-23:
OK, think I have it - I need to add the fingering to the notes in order to get it to show up. Thank you for all your help.

I added a couple of extra if (note > statements to get the LED placement precise, thank you for all the help and advice. Keep up the great work...

closed 2020-04-23 by mlewis-uk