onlaj/Piano-LED-Visualizer #100: Bluetooth Android

bradliegek3, 2020-03-31:
Hi,

I am trying to get the project working with synthesia on Android.

I am using the midi btle app to search for midi devices. I am able to connect with Rpi-bluetooth. After that, I can see Rpi-bluetooth in synthesia.
However, when I play any notes, they won't show up in synthesia.

onlaj, 2020-04-02:
Did you follow [instructions](https://synthesiagame.com/forum/viewtopic.php?p=47541#p47541)?

vzoltan, 2020-04-02:
yes, i can confirm, unfortunately it doesnt work, so i had to buy a windows tablet for this..(hint, teclast x4 is very cheap and does the job perfectly, even more.) it was about 280 usd and no extra fee or vat, its from china, to EU)

Synth49, 2020-04-19:
Hello,

I spent hours (if not days) fighting with the Bluetooth part of the setup. It's indeed painful.
I end up with a situation where Synthesia is working with the RPi (but can't light the stripe in learning mode...which sadly is the main goal i am trying to reach)

--
I tried with 5 android Devices
- Samsung Galaxy A5 < Best result here. Synthesia receives the notes from the Piano via the RPi (stripe are lighting when i press the keys)...Synthesia can send the notes to the piano but can't light the stripe (settings properly set to Keys in Synthesia RPi-Bluetooth device).
- Pocophone (can see the RPi device in Synthesia - Synthesia notice if Bluetooth goes down, no data from and to the RPi).
- Samsung Galaxy Tab S2  (same as Pocophone)
- Samsung Galaxy Camera (Midi+BTLE is not available for this device...RPI not seeing from Bluetooth settings. I was able to pair it from the RPi in command line...But obviously not working as midi device).
- Honor Lite 10 (Midi+BTLE does not sees the "RPI-Bluetooth" but i can pair the phone from bluetooth settings ...but of course won't see it in Synthesia).
---
FWIW with the Samsung Galaxy A5 I achieve same result as with a Laptop Dell Latitude 7480 with Window 10.
(With a Lenovo T420 I am roughly at the same status as with Honor Lite 10 - I noticed that "Bluetooth LE Explorer" says "central role"  is not supported for this device so i guess it's tied to some limitation with the bluetooth implementation (??) )
---
I might give up soon and go for an USB-midi connection to Synthesia (meaning probably a USB hub required as 2 usb port only on the RPi (1 for power, 1 for keyboard....and now 1 is needed to connect to Synthesia running device).

Synth49, 2020-04-25:
The issue for me was (facepalm) not properly setting "RPi-Bluetooth" as input port in the RPI Menu.
Once done it's working (lights on from Synthesia) on the Galaxy A5 (with some input losses it seems) and properly with Windows 10 when RPI device seen in Synthesia.

stepir, 2020-06-04:
@Synth49 is it reboot persistent? I have to reset at every boot - wondering if you had to do anything special to have it persist after reboots

Synth49, 2020-06-04:
Not persistent at reboot ( issue is Synthesia Bluetooth handling i think)
With Windows10 i have to 
1- unpair the Raspberry to have Synthesia properly working (or the process is running but never displayed).
2- Launch Synthesia
3- Associate the Raspberry Pi again
4- Set "RPi-Bluetooth" as input port in the RPI Menu.
I might go for a batch to automate the windows part.

stepir, 2020-06-04:
Thanks for confirming it. At least it’s not just me 😃

closed 2020-12-23 by onlaj