onlaj/Piano-LED-Visualizer #101: Bluetooth IOs

bradliegek3, 2020-03-31:
Hi,

I am also trying to get the project working with Synthesia on ipad or mac.

When I search for bluetooth devices, I can see my piano (fp30), but I can't find the RPi.

CaptainSword123, 2020-04-26:
Have you tried resetting the bluetooth settings on your rpi? You can find it under 'Ports Settings' > 'Reset Bluetooth settings'
I can connect with my iPad and set it as Input and it the LEDs light up when I hear a Song in Synthesia, but it's not interacting with my piano.

stepir, 2020-06-04:
I don't have a full automated solution but can tell you that, at least in my case, I have to connect first synthesia to the rpi via BT. THEN I can go on the RPI visualizer menu, refresh input ports and the BT-RPi server show up in the list and can selected. The problem that I have is that it's not persistent and need to do it at every boot.

closed 2020-10-11 by bradliegek3