onlaj/Piano-LED-Visualizer #102: Bluetooth and Synthesia

Roloo38, 2020-04-01:
hello. today i added the bluetooth option but i can't select the input to bluetooth rpi, and i can't find the "finger based channel'' option in "key lights", in Synthesia.
btw, i am using a macbook pro

onlaj, 2020-04-02:
According to Synthesia's author it should work out of the box, I don't have Mac to test it myself.
Does it work for you with Windows?

Roloo38, 2020-04-02:
I don't have an windows device with bluetooth, but the problem is I can't find the rpi bluetooth option in visualiser input menu

Roloo38, 2020-04-02:
the only option I have :

Midi Through:Midi Through Port-0 14:0
Rt Midiout Client:RtMidi output 132:0
USB Prokeys 88: USB Prokeys 88 Midi 1 20:0
USB Prokeys 88: USB Prokeys 88 Midi 2 20:0

PIrKo, 2020-04-04:
i dont want to create a post so i do it here>
once installed, it updates automatically in my rasperrypi? if you make any modifications, or i have to do:
git clone https://github.com/onlaj/Piano-LED-Visualizer
if u did a commit

thx

Roloo38, 2020-04-05:
@PIrKo you need to run `git pull origin master ` in your folder where is the github clone(i think:)

PIrKo, 2020-04-05:
> @PIrKo you need to run `git pull origin master ` in your folder where is the github clone(i think:)

thx! but i dont understand what i have to do lol. there is a guide to update the script? or autoupdate?

onlaj, 2020-04-23:
You need to open terminal and go to directory where the Visualizer is located. Then you need to type 
`git pull origin master`
If it doesn't work you can delete project folder/files and clone repository again.

closed 2020-12-23 by onlaj