onlaj/Piano-LED-Visualizer #105: LEDs stay on/randomly light up

XBleader, 2020-04-06:
For some reason sometimes some LEDs stay on or randomly light up when I play on the piano - this problem has been mentioned before but the person only said he fixed it by changing the hardware connection.
I've tried different cables but nothing seems to fix the issue. :(

Any idea what may cause the problem?

Thank you very much in advance!

onlaj, 2020-04-23:
Hey. It looks like connection problem. Did you try to reconnect wires to pins?

Pianoman871, 2020-04-25:
Same problem where, but I find the solution:
The USB Charger 5V2A was a bad quality. I replaced by a better quality Charger.

XBleader, 2020-04-29:
First of all, I wish to thank you guys for your reply - I really appreciate your help and the time you took! :)

I replaced the whole setup but even though I bought everything new (even a new LED-Stripe, new RaspberryPi etc.), it still appears/doesn't work properly. :(
I reconnected the wires and replaced them as well but nothing seems to help.

Furthermore, I also bought a good charger both for the LED-Stripe and the Raspberry Pi but unfortunately, it hasn't eliminated the problem in my case.

So eventually I replaced everything except for the piano and it still doesn't work.
I'm running out of ideas. :/

I don't think it's my piano's fault as the midi records perfectly fine on my laptop when I record it.

Any more ideas what may cause the problem or what I could do to fix it?

Thank you very much, guys!

Best wishes,

Tobias

Pianoman871, 2020-04-29:
Hi Tobias, the MIDI piano must transmit only on one channel. In my case, if my piano is set to transmit on all 16 channels, I get the same random light show!

onlaj, 2020-05-04:
Can you run [tests/printmidimessages.py](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/printmidimessages.py) and show me how the output looks when you play? Also, check if it's dropping any messages.
If you replaced everything then I would suspect that the piano is the source of problem. What model is that?

XBleader, 2020-05-22:
> Hi Tobias, the MIDI piano must transmit only on one channel. In my case, if my piano is set to transmit on all 16 channels, I get the same random light show!

I tried several MIDI-Channels however unfortunately it didn't have any effect and the random lights still appear. :/
Which channel did you choose if I may ask? :)

> Can you run [tests/printmidimessages.py](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/printmidimessages.py) and show me how the output looks when you play? Also, check if it's [...]

I will do that this weekend and let you know asap!
My piano model is called: Kawai CA 78
I hope this helps - thank you very much in advance! :)

closed 2020-12-23 by onlaj