onlaj/Piano-LED-Visualizer #108: For an 61 keys piano

Pianoman871, 2020-04-22:
What is the change in the visualizer.py code we must do?

onlaj, 2020-04-23:
Hey.
You need to find all occurences of "176" in code and replace it with "122" (61 x 2). 
If after the change leds are not aligned with keys try to edit code after the comment 
> #changing offset to adjust the distance between the LEDs to the key spacing

```
if(note > 92):
            note_offset = 2
        elif(note > 55):
            note_offset = 1
        else:
            note_offset = 0
```

This code means that after led `#92` (counting from left) the offset should be set to 2, so instead lighting led `#92` it will light led `#94`. I'ts the same with notes over `#55`, but the offset value is 1.

I'm planning to add an option to change those values in Visualizer settings.

Pianoman871, 2020-04-23:
Solved! Solution provided by: mlewis-uk <notifications@github.com>:
For an 61Keys keyboard, you need to edit the visualizer.py script, under the LED offset section and change it to look like the below:
if(note > 72):
note_offset = 33
elif(note > 35):
note_offset = 32
else:
note_offset = 31

Thank you mlewis-uk!

closed 2020-04-23 by Pianoman871