onlaj/Piano-LED-Visualizer #109: Bluetooth Midi signal

saltimbranco, 2020-04-23:
I can get BT connection between the RasPi and the PC. Synthesia can send signal to light up the LED but there is no piano Midi signal send to PC. Any suggestion? Thanks a lot.

onlaj, 2020-04-23:
Hey. Did you configure input in Synthesia settings?

saltimbranco, 2020-04-23:
Yes I did.

CaptainSword123, 2020-04-26:
same problem, LED is working fine, I can also see the midi bluetooth server but get no signal after connecting. I wrote an e-mail to neuma studios over a month ago but never recieved a reply:/

saltimbranco, 2020-05-05:
I found that if I run connectall.rb it shows

/usr/local/bin/connectall.rb: line 5: t: command not found
/usr/local/bin/connectall.rb: line 6: ports: command not found
/usr/local/bin/connectall.rb: line 8: syntax error near unexpected token `\d*'
/usr/local/bin/connectall.rb: line 8: `  /client (\d*)\:/=~l'

is this causing the problem?

rickstaa, 2020-06-11:
@saltimbranco You can solve that error by changing the shebang from `#!/usr/bin/ruby` to `#!/usr/bin/env ruby`. I suspect it has something to do with the way ruby has been set up inside your system and that `#!/usr/bin/env ruby` uses the PATH which points to the correct ruby version.

HangryPanda, 2020-10-23:
I have similar issues on my Surface Pro 4 (Windows). Are you able to see the 'RP Bluetooth' option in the 'Input' section of the Pi? I can see that the Pi is paired to my Surface, and I can set the input settings in Synthesia to my Pi via bluetooth, same with the output, but for some reason when I look at the Port Settings on the Pi I don't see the RP4-Bluetooth option. I reset the BL settings and then refresh ports and sometimes that works. Other times I have to SSH from another terminal window and that seems to do something to the PI that enables or refreshes something that causes the pi to register the RP4-Bluetooth as an option. 

Maybe someone smarter than me can look at the code and see what causes the bluetooth service stall or hang up - if that's even a thing. :p I hope this helps you and someone can patch it :p

rickstaa, 2020-11-03:
@HangryPanda After I changed the shebang to `#!/usr/bin/env ruby` everything worked fine. Maybe you can tryout it out using the iso file I'm currently using (see #15)?

kmbvip, 2020-11-25:
@rickstaa please help :). I have Your IMG of piano led visualizer. Piano led working fine, everything shining and works. When i start RPI i see that everything works good, blotooth midi is ON. On windows when i pairing RPI i see in system that there is MIDI IN and MIDI OUT but ..... synhtesia didnt see it.

I tried everything like You write but it didnt work. I have couple of MIDI controller and when i connect it to PC sythnesia see it and wok fine. But not with RPI MIDI. What i have to do?

rickstaa, 2020-11-25:
Hey Kmbvip,

I unfortunately did not thoroughly experiment with the RPI Bluetooth feature. I tried it with Synthesia several times but more than often I must restart it multiple times before it works. I did not yet find what causes it to work sometimes and causes it to fail other times. In the end I just gave up and used a cable. I am however able to successfully connect it to the Bluetooth of my Roland lx-17. I think if you really need to have to Issue solved it is best to create a detailed issue report with the errors yourexperiencing. By doing this one of the original authors might be able to help you out.

Sorry if I couldn't be of more help,
Best wishes,
Greetings,
Rick

kmbvip, 2020-11-25:
There is no problem, thanks for the answer.

Ok so one more question. How do You connect it with cable? One port is
going to keyboard to take a MIDI that LED can shine, and second is for a
power.

rickstaa, 2020-11-26:
> There is no problem, thanks for the answer.
> 
> Ok so one more question. How do You connect it with cable? One port is
> going to keyboard to take a MIDI that LED can shine, and second is for a
> power.
> 
> 
> 
> 
> 
> 
> > Hey Kmbvip,
> >
> > I unfortunately did not thoroughly experiment with the RPI Bluetooth
> > feature. I tried it with Synthesia several times but more than often I must
> > restart it multiple times before it works. I did not yet find what causes
> > it to work sometimes and causes it to fail other times. In the end I just
> > gave up and used a cable. I am however able to successfully connect it to
> > the Bluetooth of my Roland lx-17. I think if you really need to have to
> > Issue solved it is best to create a detailed issue report with the errors
> > yourexperiencing. By doing this one of the original authors might be able
> > to help you out.
> >
> > Sorry if I couldn't be of more help,
> > Best wishes,
> > Greetings,
> > Rick
> > From: kmbvip <notifications@github.com>
> > Sent: 25 November 2020 17:33
> > To: onlaj/Piano-LED-Visualizer <Piano-LED-Visualizer@noreply.github.com>
> > Cc: Rick <rick.staa@outlook.com>; Mention <mention@noreply.github.com>
> >
> >
> > @rickstaa<
> > https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Frickstaa&data=04%7C01%7C%7C8131660ea17d4f594cd608d8915fc752%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637419187832571710%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=7vQDiAULHcQn03uTmx%2FzTAHjbzlwhjHdZub6IQ6IX%2F0%3D&reserved=0>
> > please help :). I have Your IMG of piano led visualizer. Piano led working
> > fine, everything shining and works. When i start RPI i see that everything
> > works good, blotooth midi is ON. On windows when i pairing RPI i see in
> > system that there is MIDI IN and MIDI OUT but ..... synhtesia didnt see it.
> >
> > I tried everything like You write but it didnt work. I have couple of MIDI
> > controller and when i connect it to PC sythnesia see it and wok fine. But
> > not with RPI MIDI. What i have to do?
> >
> > https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fonlaj%2FPiano-LED-Visualizer%2Fissues%2F109%23issuecomment-733815014&data=04%7C01%7C%7C8131660ea17d4f594cd608d8915fc752%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637419187832581705%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=oSuzRDNDzn5pLxCzXdKo9zcfkJpvs7bkEHbNX34q0X8%3D&reserved=0>,

If I remember it correctly I connected to my piano with bleutoorh. You can however also use a [USB Y cable ](https://m.alibaba.com/product/60357498922/Micro-USB-Splitter-cable-1-male.html?spm=a2706.7843667.1998817009.12.20cd4921jUZK3N&__detailProductImg=%2F%2Fs.alicdn.com%2F%40sc01%2Fkf%2FHTB1BCUjdCtYBeNjSspaq6yOOFXaT.jpg_140x140xz.jpg). I currently do not have the time to investigate why the bleutooth connection to synthesia is broken. Maybe somebody else knows.

closed 2020-12-23 by onlaj