onlaj/Piano-LED-Visualizer #110: On an 61Keys keyboard, the first MIDI note is "C2"

Pianoman871, 2020-04-23:
Thank you for the update code, but the first MIDI note do not light up the first LED on the strip.
We need an update code to reflect that on a 61keys keyboard, the first MIDI note "C2" will light up the first LED on the strip.
Details:
On an 88Keys keyboard, the first left MIDI note is A0 but on an 61Keys keyboard the first MIDI note is C2.
We need an update code to reflect that. 
Any solution?
Thank you,

onlaj, 2020-04-23:
https://github.com/onlaj/Piano-LED-Visualizer/issues/108#issuecomment-618303147

mlewis-uk, 2020-04-23:
I found that the below worked for me. You need to edit the visualizer.py script, under the LED offset section and change it to look like the below:

if(note > 72):
    note_offset = 33
elif(note > 35):
    note_offset = 32
else:
    note_offset = 31

Pianoman871, 2020-04-23:
Solved! It work great! Thank you mlewis-uk!
For an 61Keys keyboard, you need to edit the visualizer.py script, under the LED offset section and change it to look like the below:
if(note > 72):
note_offset = 33
elif(note > 35):
note_offset = 32
else:
note_offset = 31

closed 2020-04-23 by Pianoman871