onlaj/Piano-LED-Visualizer #111: Skipping LED's

cercyo, 2020-04-29:
Hello!
First of all thank you for this wonderfull project!
Could you please so kind to help with my issue?

My piano is Yamaha P-45 (88 keys)
My LED-strip is WS2812B 144 diods  per metr (http://ledreklamashop.ru/goods/Adresnaya-svetodiodnaya-lenta-RGB-WS2812B-5-Volt-SMD5050-144-LED-LUX-class?mod_id=213608059#show_tab_3)

The first key on the piano turns on the third diode
The second key on the piano turns on the fifth diode

Why it isn't start from first diode and don't turn 2 didods instead of one?

[image]

[image]

onlaj, 2020-04-30:
Hey, I designed it this way. If you want to light more keys with one key press you can set in visualizer settings that it will light 3 leds.

cercyo, 2020-05-01:
Thank you! It's perfect for me!

closed 2020-05-01 by cercyo