onlaj/Piano-LED-Visualizer #113: Midi files are still to fast

eengamer, 2020-05-02:
Hello onlaj, after hours of editing i came to the realisation that the midi still is to fast on the starr it was in sync for a long time but later in the video it isint synced at all. I tought you sad that this problem was fixed but it clearly isint. Now all my work is for nothing :(

onlaj, 2020-05-02:
I tested 8 minutes long MIDIs and they were fine, how long is your midi?

eengamer, 2020-05-02:
Mine is around 7 minutes but it isint right i updated the pi yesterday and recorded today but somehow i dunno why

eengamer, 2020-05-02:
If you want i can send you the bottom recording and top recording and wav

onlaj, 2020-05-02:
Sorry to hear that.
You can try to set higher tick_per_beat
Change 10000 in [this line](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L1329) to 20000
and 20000 to 40000 in lines [1356](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L1356) and [1358](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L1358)

btw. how fast do you play? If it's some Chopin's Etude compilation then I'm not suprised :D

eengamer, 2020-05-02:
I have found a solution that in gonna try instead of recording the sound on the oiano iam going to record midi on my piano and use a vst with it for the sound. This time it was not choping it was my own version of nuvole. Bit i was planning on doing fantasie impromptu one day

closed 2020-05-04 by eengamer