onlaj/Piano-LED-Visualizer #114: Color fade between to colors

eengamer, 2020-05-04:
Hello, i was wondering if there is an way you can add a feature that the leds go from 1 chosen color amooth to the other color. Like in the picture i pinned to this one.
[image]

eengamer, 2020-05-10:
If you do this. When will this come out?

vzoltan, 2020-05-18:
you can play with the settings, where you can set the rainbow effect, 

mine is set to -15 30 0
and that gives a nice smooth gradient from deep red to greenish yellow, from low to high, left to right.

i dont rememeber wher is it exactly in the menu, but in the settings.xml its like this:

```
<fadingspeed>10</fadingspeed>
<color_mode>Rainbow</color_mode>

<rainbow_offset>-15</rainbow_offset>
<rainbow_scale>30</rainbow_scale>
 <rainbow_timeshift>0</rainbow_timeshift>
```

eengamer, 2021-05-18:
am trying it right now but somejow it is only staying on blie and there is no gradient even tho i pit in the colord

onlaj, 2021-05-18:
Works for me, did you update all files?
[image]

eengamer, 2021-05-18:
i think i have to rewire my wires. they got loosen up a bit and thats what causing it to not function that well. or my powersupply may have a fault

onlaj, 2021-05-18:
If it doesn't help we can try to debug it.
Inside [this](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L2337) method add line with

`print ("position: "+str(position)+" leds: "+str(ledstrip.led_number)+" start: "+str(self.gradient_start["red"])+"end: "+str(self.gradient_end["red"]))`

run modified code, try to play something and copy paste here output from console.

closed 2021-08-16 by onlaj