onlaj/Piano-LED-Visualizer #115: Problem by running visualizer

JonasAndSchmidt, 2020-05-05:
Hello,
I have the problem, that I am not able to open the file 'visualizer.py´. I guess the problem is that I am running Python 2.7 and I am not able to install one of the requirements (python-rtmidi). I looked online and I need to habe Python3 to run the pip. So I installed Python3 but I am not able to connect with the RaspberryPi in Python3. Can someone help me?

onlaj, 2020-05-06:
Did you install everything as follows?

```
sudo apt-get install -y autotools-dev libtool autoconf
sudo apt-get install -y libasound2-dev
sudo apt-get install -y libusb-dev libdbus-1-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev
```

After that try to install python-rtmidi

JonasAndSchmidt, 2020-05-06:
Thank you, I tried that. 
It's working now after I switched to Windows with Bootcamp. I started all over again and did the same things but it's working now. I still don't know why it didn't work but it doesn't matter anymore.

closed 2020-05-06 by JonasAndSchmidt