onlaj/Piano-LED-Visualizer #118: Bluetooth, Synthesia and macOS won't work correctly in some cases and the fix

darkfall, 2020-05-11:
This is not a real "issue" but I'm putting my investigation result for the record.

My setup: Yamaha, rPi 4, macOS 10.15.4, Synthesia.
 
It works fine if there is only one key press / release at a time, but if there is more the "note" and "velocity" fields of the events sometimes gets messed (looks like swapped) due to a parser bug in the BLUEZ driver stack mentioned in the tutorial (the fork from oxesoft).

I forked it and have a fix: https://github.com/darkfall/bluez/commit/b02feada9bf934d9b2136fbeebbf97d275b5e080. If you have this kind of issue you can try compiling my fork instead of the one mentioned in the tutorial. Though I haven't tested the code in other setups (like windows), but in theory it should be safe.

**Background:**
macOS (maybe iOS or Apple midi driver in general) coalesces note events if there are multiple happening at the same timestamp (midi spec does not mention this maybe I missed something), i.e.

      0x90, NOTE1, VELOCITY1, NOTE2, VELOCITY2, NOTE3, VELOCITY3

But the parser in bluez stack (libmidi.c) has a bug in processing the timestamp that in this case, NOTE2 will be treated as a timestamp and skipped in the event. Thus in this case, the Visualizer will receive: [NOTE1, VELOCITY1], [VELOCITY2, NOTE3]... and cause all sorts of problems due to mismatching events and random lights on / off due to velocity treated as a note.

stepir, 2020-05-24:
hi @darkfall and thanks - I'm still investigating as I have other issue with my setup but I think I'm seeing the same issue as well. I'm on IOS (Synthesia running on IPAD and connecting over BT to the PI). Will report back later

stepir, 2020-05-24:
@darkfall 

please look at this video - do you think it's the same issue you described?

https://youtu.be/rg4i7rZvI7M

stepir, 2020-05-24:
Hey @darkfall 

had some spare time and realized testing would not have been such a big risk - I gave it a try and recompiled bluez using your fork: worked like a charm.

Thanks a lot!!!!!

darkfall, 2020-05-24:
Glad it worked for you!

closed 2020-12-23 by onlaj