onlaj/Piano-LED-Visualizer #119: Support for pianos with less than 88 keys

onlaj, 2020-05-14:
In the latest [commit](https://github.com/onlaj/Piano-LED-Visualizer/commit/d4a1a3dd96baebe69db43c4a80bd96fb2201d112) I added an option to change number of leds using Visualizer menu.
Under "LED Strip Settings" submenu you will find "Led count", which value is twice the number of your piano keys. The second new option is "shift" which allows you set an offset, useful when you want, for example, [start on first pixel/LED not the 3rd](https://github.com/onlaj/Piano-LED-Visualizer/issues/117) 
I have only one piano so I'm not able to test this update properly. This thread is made for reporting problems and asking questions related to this topic.

Berrance, 2020-05-14:
Thanks for the response, not had chance to try this yet. I will probably be able to check over the weekend. Our Piano is 88 keys and a Kawai ES110 so virtually the same as yours. The first 2 LED's weren't a problem until pulling them in the LED profile.

Hope I am wrong but if the "shift" option just changes the value of "note_offset" I don't believe it will work as I already tried this. I have already changed "note_offset" to 2 and it did make the visualiser start on the first led but it messed with the key spacing for the rest of the keys. If the "shift" option is new and litrally moves the LEDs one way or the other a specified amount of LEDs than I believe it will be good.

I will let you know what the result is and many thanks for you good work

Berrance, 2020-05-17:
Excellent work my friend. Works perfectly.

Many thanks

closed 2021-08-16 by onlaj