onlaj/Piano-LED-Visualizer #121: Midi Files

harryG15, 2020-05-19:
Hi, just wondering if it’s at all possible to record midi on the Raspberry Pi? And then either save it onto a usb from the device?
Or WHILE ITS PLAYING - record it into my computer Live??
Let me know your thoughts!
Thanks
Harry

vzoltan, 2020-05-19:
yes it is.

i did it on my setup you need to add these lines to your [menu.xml at line 207-212 here](https://github.com/vzoltan/Piano-LED-Visualizer/blob/master/menu.xml#L207)

```
        <menu text="Export MIDI">
                <Export_MIDI text="Confirm"></Export_MIDI>
        </menu>
        <menu text="Import MIDI">
                <Import_MIDI text="Confirm"></Import_MIDI>
        </menu>
```

and to your visualizer.py at line [1063-1079 here](https://github.com/vzoltan/Piano-LED-Visualizer/blob/master/visualizer.py#L1063)

```
        if (location == "Export_MIDI"):
            if (choice == "Confirm"):
                menu.render_message("", "Exporting MIDI files...", 5000)
                call("sudo mount /dev/sda1 /media/import", shell=True)
                call("cp /home/pi/Piano-LED-Visualizer/Songs/* /media/import/midi/", shell=True)
                call("sudo umount /dev/sda1 /media/import", shell=True)
            else:
                self.go_back()

        if (location == "Import_MIDI"):
            if (choice == "Confirm"):
                menu.render_message("", "Importing MIDI files...", 5000)
                call("sudo mount /dev/sda1 /media/import", shell=True)
                call("cp /media/import/midi/*.mid /home/pi/Piano-LED-Visualizer/Songs/", shell=True)
                call("sudo umount /dev/sda1 /media/import", shell=True)
            else:
                self.go_back()
```

and then you need to create a folder on your usb pendrive to the ROOT dir, named to "midi"

when you have files on your pendrive in the midi folder and you select import on the RPI in the visualizer menu, it will copy all your midi files from the usb to the pi, when you select EXPORT it will copy all the midis from the pi to the usb.

harryG15, 2020-06-23:
Ok thank! Tried adding it, but program wouldn’t start due to an error,
What’s the best way to add the code in?
I’ve never done it before!! I’ve only ever cloned the respository
Would appreciate any help :)

closed 2020-06-23 by harryG15