onlaj/Piano-LED-Visualizer #123: Feature request. Split midi tracks when recording in multicolor mode.

sergiusmuzzz, 2020-05-26:
I'll explain. 
I want to be able to match notes colors of synthesia with LED colors when playing in multicolor mode. 
To do so midi track needs to be split into as many tracks as there are colors I set in multicolor mode. All notes in each track need to be painted in video editing software accordingly.
I've tried to do it manually by duplicating midi tracks and removing extra notes from each track to leave only necessary notes for each color and if I have more then 2 colors it's already turns into a nightmare. It's way too tedious process for a human being.
I suspect it's possible to do it automatically by developing and adding option that allows to record midi track for each color in multicolor mode. I know it might be not that easy but it would be awesome if you @onlaj could do it! Because it's soo beatuful! 
Here is one of my examples. https://drive.google.com/open?id=1YxxnQC7rwMOEFCqbaKg4pB4Z4DN2KG0Q

onlaj, 2020-05-26:
Hey, it's already on my to-do list, but thanks for letting me know that this feature is needed :)

sergiusmuzzz, 2020-10-08:
Is there plans to implement this any time soon? :)

onlaj, 2020-10-08:
Soon™
I will try to make it in next update :)

sergiusmuzzz, 2020-10-08:
Awesome!

onlaj, 2020-10-09:
@sergiusmuzzz Should I save control change (pedals) data on separate tracks or just keep them on main track (which contains all notes)? What do you think?

sergiusmuzzz, 2020-10-09:
@onlaj  Whatever is easier to implement. I don't think it would be necessary for pedals to be on separate tracks.

onlaj, 2020-10-12:
I commited changes and pinned this issue. Use this thread to report any problems related to recording in multicolor mode.

sergiusmuzzz, 2020-11-15:
It works! Here is the [result](https://youtu.be/481AKTIWc7o).
Huge thank you for this feature and for the app in general!

The only think is that when I import midi tracks of each color into DAW they aren't aligned with each other properly and start from their first note. It would be awesome if somehow midi tracks would initially get imported at expected place. What if to compare first notes of each colored track with first notes of main track till all track were checked: 
Here is how I assume condition would look like in JS synaxis:
```
//Only need to check the beginning of the song till the first notes of all colored tracks passed check
for(let i = 0; i < colorTracksArr.length; i++){
     if (thisColorTrack.firstNote !== mainTrack.currentNote){
           addPlaceholderNoteToThisColoredTrack(); // write a note(any, C0 for e.g.) to allow a real first note to be at the correct 
     place (all those extra placeholder notes will be removed once track is imported)
     }
}
```

I'm sure you'll have couple more better solutions. But if it's not so easy no problem at all I'm already soo happy it works as it is! :-)

And if it's not clear what I mean I can provide more details.

Here is a small manual how to align colored midi tracks manually. Import main midi track to use as a guideline. In your DAW 
 snap cursor to the first note in main track and then snap track of corresponding color to cursor and then find position of the first note for the next track in main track and snap cursor there and then move next track to cursor at new position and so forth. Sounds a bit complicated but it gets easier with  practice. I also can send screenshots if someone will need to understand the workflow better.

darmap, 2021-01-16:
sorry I still don't understand, can you send a screenshot

closed 2021-08-16 by onlaj