onlaj/Piano-LED-Visualizer #124: port input not persistent? +other issues

stepir, 2020-05-28:
Hi

I'm trying to sort some misbehaving I'm seeing when Visualizer is configured to enable "next-to-play keys".

At every boot I have to manually reconnect my iPad via BT - then refresh the input ports and associate again the 'RPI Bluetooth" to input. Is that by design? or it should be the case? 
Perhaps I can work on the BT stack to auto connect my IPAD at boot but I'm still wondering if the rest need to be done at every boot?

The other issues I'm seeing are
- Visualizer screen is not always shown, although I'm seeing python running. It will always be shown if I reboot after the initial boot
- The screensaver exit immediately, even when I run it manually - like if a key is immediately pressed (which is not the case). Any clue at what I could look at?
- The left and right hand led are shown of the same color, actually even when I'm learning/playing only right hand - the led of the left are automatically lighted (same color). Is that correct? if not what it could be?
- when I have 3 notes in rapid sequence (e.g.) only the led of the first note is lighted.

Perhaps I should produce a video of the last two issue...

onlaj, 2020-05-28:
> At every boot I have to manually reconnect my iPad via BT - then refresh the input ports and associate again the 'RPI Bluetooth" to input. Is that by design?

No, it doesn't work like that for me, but I use Windows PC. Once I connect my tablet to RPi it always reconnect correctly.

>Visualizer screen is not always shown, although I'm seeing python running. It will always be shown if I reboot after the initial boot

What do you mean by initial boot?

> The screensaver exit immediately, even when I run it manually - like if a key is immediately pressed (which is not the case). Any clue at what I could look at?

Screensaver exits when you press KEY2 or there is incoming midi message from input. If they KEY does not behave incorrectly under other conditions (going back randomly while browsing menu) then I suspect your piano is sending some midi messages.

> The left and right hand led are shown of the same color, actually even when I'm learning/playing only right hand - the led of the left are automatically lighted (same color). Is that correct? if not what it could be?

I can't see that on video, did you choose "finger based" option in key light settings?

> when I have 3 notes in rapid sequence (e.g.) only the led of the first note is lighted.

I'm guessing this is caused by bluetooth connection, although I don't have this problem on my setup.
If you don't use Synthesia on Windows PC then I suggest buying this device: http://compasflamenco.com/midi-c-3/midi-usbusb-p-4.html wired connection should be far more reliable than bt.

stepir, 2020-05-28:
With initial boot I mean when I initially turn on the raspberry 

The other menu item works just fine and I noticed now that also the screensaver start working after many many minutes

Probably you are right there is still some incoming midi traffic happening.

I can’t use a usb to usb as my piano doesn’t have a USB port. I’m using now a midi to USB adapter connected to my raspberry. Would it be an option to connect it to my iPad using a lightning to USB adapter? Or should I use the device you advised in the middle (so modo to usb - usb to usb - usb to lightning)?

Thanks!

stepir, 2020-06-04:
I've started from scratch and reinstalled once again. Visualizer not starting was related to auto login and it's fixed.

Something was messed with BT and now it's working much better - I don't have the issue with led not being turned on with rapid sequences.

Still I have to manually connect the RPI in Synthesia settings at every boot (connecting from the BT IOS panel is not needed anymore), refresh the input ports on the RPI and set it to BT.

@onlaj the issue I have now, and wanted to check if you have idea on where to start looking at, is that right hand notes are not sustained. When I press that note the sound is reproduced only for a fraction of time - instead until I keep pressing the key. I can workaround it by pressing the sustain pedal. It only happen with the right hand notes, the left one are working fine!! Do you have any clue?

Thanks again!!

onlaj, 2020-06-04:
> Still I have to manually connect the RPI in Synthesia settings at every boot (connecting from the BT IOS panel is not needed anymore), refresh the input ports on the RPI and set it to BT.

I know this is not a solution to your problem, but maybe just don't shutdown your RPi? I keep mine on for over a year now.

> right hand notes are not sustained
What do you mean by that? The leds are turning off even if you keep pressing the key?

stepir, 2020-06-04:
Led are working properly but only right hand notes are played for a fraction of second doesn’t matter how long I press the relative key.

The left hand notes/keys work properly, the note is played depending for how long I press the key.

stepir, 2020-06-04:
> I know this is not a solution to your problem, but maybe just don't shutdown your RPi? I keep mine on for over a year now.

Led are working properly but only right hand notes are played for a fraction of second doesn’t matter how long I press the relative key.

The left hand notes/keys work properly, the note is played depending for how long I press the key.

onlaj, 2020-06-04:
So it happens only when you play with Synthesia, right? I think it has nothing to do with Visualizer,

stepir, 2020-06-04:
> So it happens only when you play with Synthesia, right? I think it has nothing to do with Visualizer,

it only happens when I use Visualizer - if I configure in Synthesia to use the iPad as the output, it work properly

stepir, 2020-06-04:
see here:

first with Visualizer as output (hope you can see the settings in the video) - then with iPad as the local output

https://streamable.com/9iolac

I'm also doing a video with the left hand (Visualizer as output) where you'll see that notes are sustained depending on how long I press the key

stepir, 2020-06-04:
here is a video with the same settings used also for the right hand - this time I'm only learning the left hand notes, as you can see notes are sustained depending on how long I press the key:

https://streamable.com/gvesbw
https://youtu.be/aIJImTT0lGs

onlaj, 2020-06-04:
Oh wow, this is really weird and unexpected behavior. Do you have other devices, such as Windows PC to test it in different conditions? I would also try to change some Syntesia settings, maybe try to set "local notes" to on?

stepir, 2020-06-04:
with local notes on/off the behavior won't change - I don't think it's related to the iPad as it's a close device with no way of tricking it. What is unexpected is that the left hand work properly.

I can test it on my Mac

stepir, 2020-06-04:
@onlaj I'm very sorry for wasting your time for issue non related to Visualizer. I just realized by testing it on my mac that it's an entire set of songs that I have converted from video to midi using an app. 

Visualizer works really well now! again sorry!!!!!!!!!!!!

onlaj, 2020-06-04:
Not a problem, I'm glad you figured it out.

stepir, 2020-06-05:
Closing this as it seems that there is no way to handle it as of now

closed 2020-06-05 by stepir