onlaj/Piano-LED-Visualizer #126: Piano Visualizer & Synthesia

MarcoNoah, 2020-06-03:
Thank you for your project.  Apologies in advance if this is not the correct forum to raise this. I have no programming experience   I have set up on a Rasberry pi zero the visualiser works with my Keyboard Roland FP30 via USB cable.

I would like to use Synthesia.  I have managed to connect to my Ipad via bluetooth.  When using freeplay keys pressed on the FP30 are mirrored on the Ipad and you can see a colour going up the screen showing the note pressed. 

If I play a music from Synthesia the sound comes out on the FP30 but the LED lights on visualizer  does not light up.  Have I done something wrong?  

I would be grateful for any help . 

I set up also on a rasberry pi4 but the same thing happens.  I was hoping to connect my laptop to one USB port on the RPI4 and the other to the FP30.  My laptop didn'tpick up that it was connected to the RPi4

Sorry if this is a stupid question but I thought I would bite the bullet ask as I want to undersand more

onlaj, 2020-06-03:
> If I play a music from Synthesia the sound comes out on the FP30 but the LED lights on visualizer does not light up. Have I done something wrong?

If you want your iPad to control led strip you have to select it as input under "Ports Settings" menu.

For a cable connection you need to buy this: http://compasflamenco.com/midi-c-3/midi-usbusb-p-4.html

btw. you managed to make Synthesia works via bluetooth? Many users reported problems with that setup. What iPad do you have and what steps did you make to achieve that? I don't have Apple products myself to test it.

MarcoNoah, 2020-06-03:
I managed to connect the ipad to the RPI see attached picture

 

 

Issue is now the led lights are not working and the FP30 is making some strange noises.

stepir, 2020-06-04:
Marco it's better if you interact using GitHub web interface and not just by replying to the notification emails you receive. The attachments photo are lost. click below where you see 'view on github'

MarcoNoah, 2020-06-05:
Can you advise if you purchase the usb midi device what the connections would be between the RPI the laptop/IPad and the  Piano

stepir, 2020-06-08:
Marco it would be:

Piano Midi to USB -> Raspberry -> MIDI USB -> IPAD

Given the PI Zero ha only 1 USB port (the second is only for power) you would need an USB OTG adapter or a USB hub

MarcoNoah, 2020-06-08:
Thanks.

 

I managed to set up experimenting .  I used the RPI 4 so it has additional usb ports

 

Do you know if it is possible to use a touch screen  with the PI4 for the visualizer?

stepir, 2020-06-09:
as long as you use the same display

MarcoNoah, 2020-06-09:
Sorry you have confused me  when you same use the same display – does this mean the touch screen wouldn’t work?

stepir, 2020-06-09:
the code here support only the 1.44" display

closed 2020-12-23 by onlaj