onlaj/Piano-LED-Visualizer #127: Bluetooth IOS Ipad issue

SkoomaAbuser, 2020-06-05:
Hi, really love this project
! managed to get it all working tonight, just struggling with connectting my ipad to the raspberrypi. When i go to synthesia-bluetooth settings, i am able to find my rpi bluetooth,  it tries to connect then just stops.

any help would be appreciated.

stepir, 2020-06-05:
Hi

been into it as well - would be worth looking at the output of:

sudo systemctl status bluetooth

mine was like 

pi@RPi-Clavinova:~ $ sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-06-05 09:23:57 BST; 2min 32s ago
     Docs: man:bluetoothd(8)
 Main PID: 420 (bluetoothd)
   Status: "Running"
   Memory: 1.9M
   CGroup: /system.slice/bluetooth.service
           └─420 /usr/libexec/bluetooth/bluetoothd

Jun 05 09:23:51 RPi-Clavinova bluetoothd[420]: Starting SDP server
Jun 05 09:23:51 RPi-Clavinova bluetoothd[420]: Bluetooth management interface 1.14 initialized
Jun 05 09:23:51 RPi-Clavinova bluetoothd[420]: Failed to set mode: Busy (0x0a)
Jun 05 09:23:51 RPi-Clavinova bluetoothd[420]: Failed to set privacy: Rejected (0x0b)
Jun 05 09:23:57 RPi-Clavinova systemd[1]: Started Bluetooth service.
Jun 05 09:24:52 RPi-Clavinova bluetoothd[420]: No cache for 40:C9:45:B5:BF:BC
Jun 05 09:25:33 RPi-Clavinova bluetoothd[420]: No cache for 40:C9:45:B5:BF:BC
Jun 05 09:25:40 RPi-Clavinova bluetoothd[420]: No cache for 40:C9:45:B5:BF:BC
Jun 05 09:25:49 RPi-Clavinova bluetoothd[420]: No cache for 34:AB:37:AB:45:19
Jun 05 09:26:03 RPi-Clavinova bluetoothd[420]: No cache for 34:AB:37:AB:45:19

and I solved by increasing ExecStartPre=/bin/sleep 6 to ExecStartPre=/bin/sleep 60

in sudo nano /lib/systemd/system/btmidi.service

now it shows

pi@RPi-Clavinova:~ $ sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-06-05 09:32:43 BST; 15min ago
     Docs: man:bluetoothd(8)
 Main PID: 408 (bluetoothd)
   Status: "Running"
   Memory: 2.4M
   CGroup: /system.slice/bluetooth.service
           └─408 /usr/libexec/bluetooth/bluetoothd

Jun 05 09:32:42 RPi-Clavinova systemd[1]: Starting Bluetooth service...
Jun 05 09:32:42 RPi-Clavinova bluetoothd[408]: Bluetooth daemon 5.43
Jun 05 09:32:42 RPi-Clavinova bluetoothd[408]: Unknown group GATT in main.conf
Jun 05 09:32:43 RPi-Clavinova bluetoothd[408]: Starting SDP server
Jun 05 09:32:43 RPi-Clavinova systemd[1]: Started Bluetooth service.
Jun 05 09:32:43 RPi-Clavinova bluetoothd[408]: Bluetooth management interface 1.14 initialized

Other things that might be helping a quick connect is manually pairing and trusting your IPAD from the PI using 

sudo bluetoothctl -a
pair XXXXXX
trust XXXXX

where XXXXX is the mac address of your iPad

SkoomaAbuser, 2020-06-06:
I will try this tonight. thank you.

SkoomaAbuser, 2020-06-07:
Managed to get it working tongiht.

ive run into another issue though, everyting was running smoothly, then suddenly synthesia started to recognise the notes i was playing wrong. Synthesia seemed to think that i was pressing the note to the left of the one i was actually playing. It happened when i was changing some settings in the visualizer, is it something ive changed?

stepir, 2020-06-08:
are you sure about the MAC address used?

SkoomaAbuser, 2020-06-09:
im pretty sure yes, the pi is connected to the pi when i go to my bluetooth settings i ios, it seems to be synthesia thats having trouble connecting. I think thats what it seems to be anyway.
[image]
[image]

im also getting this in the sudo bluetoothctl 

[Christopher’s iPad]# trust
Changing  trust succeeded
[Christopher’s iPad]# pair
Attempting to pair with
Failed to pair: org.bluez.Error.AlreadyExists
[Christopher’s iPad]#

it paired with a passcode. Just cant get it to connect within synthesia,

stepir, 2020-06-09:
looks good this way - Synthesia have issue in connecting first time for me as well. Normally I kill the app reopen and it connect right away. 

Just to make it clear with delayed to 60 and manual pairing and trust, a lot of time I have synthesia even connecting automatically without doing anything.

Sometimes (50% or less) - I open Synthesia and the PI is not connected and won't connect. Then I kill the App reopen and will connect immediately.

You might want to try to increase 

ExecStartPre=/bin/sleep 6

to

ExecStartPre=/bin/sleep 60

into /lib/systemd/system/btmidi.service

SkoomaAbuser, 2020-06-12:
Thanks for the help, ive managed to get conncted pretty frequently now. 

Would you mind sharing your synthesia settings and your visuliser settings? im now habing another issue with the lights and ssynthesia, only some of the notes light up, and all the lights are white, i have no green or blue.

stepir, 2020-06-12:
I'm glad it was helpful!

unfortunately I think we will need @onlaj help then - I'm having the same issue, while I have somehow clear what settings should be used in Synthesia, there are some in Visualizer which I don't understand: e.g. skipped notes normal, finger based or none?

In Synthesia make sure you have my notes on, key lights set to finger based - rest off

onlaj, 2020-06-12:
@stepir abouty skipped notes:
https://github.com/onlaj/Piano-LED-Visualizer/issues/85
https://github.com/onlaj/Piano-LED-Visualizer/commit/e72f6f1afa29c80ed64f0015372a5cc2c30a1281
While playing with Synthesia assist you should set skipped notes to normal, then it will only light keys with blue and green color.

HangryPanda, 2020-10-20:
I'm having similar issues, but I can't get the keys to light up in Synthesia assist. I'm also experiencing a lot of notes that won't play or all notes from the left hand won't play at all. I don't have any issues with establishing a Bluetooth connection either. 

Settings: 
- Input: RPi Bluetooth
- Output: RPi Bluetooth - my notes - key lights - finger based - all other settings off 

I ordered the Sevilla MIDI-USB-USB connector, but it's not going to arrive for a while and I'm hoping I can get it working. I've also tried darkfall's solution and that didn't help.

stepir, 2020-10-20:
> I'm having similar issues, but I can't get the keys to light up in Synthesia assist. I'm also experiencing a lot of notes that won't play or all notes from the left hand won't play at all. I don't [...]

in my case it was just a waste of money - with the midi-usb-usb I had exactly the same behavior I had with BT

HangryPanda, 2020-10-20:
> in my case it was just a waste of money - with the midi-usb-usb I had exactly the same behavior I had with BT

I'm hoping my experience goes a little differently - it's not a cheap solution by any means - I think I paid like $60 with shipping for it. 

Were you able to find a fix for the BT?

onlaj, 2020-10-20:
> in my case it was just a waste of money - with the midi-usb-usb I had exactly the same behavior I had with BT

@stepir What do you mean by that? Does it drop connection or just notes?

stepir, 2020-10-21:
> > in my case it was just a waste of money - with the midi-usb-usb I had exactly the same behavior I had with BT
> 
> @stepir What do you mean by that? Does it drop connection or just notes?

drop notes - I've described the issues here:
https://github.com/onlaj/Piano-LED-Visualizer/issues/128

closed 2021-08-16 by onlaj