onlaj/Piano-LED-Visualizer #129: Create pipy package and setup.py file

rickstaa, 2020-06-11:
@onlaj First of all, thanks a lot for creating this repository it works very well! I was creating something similar when I found your code. @onlaj Maybe it is an idea to create a pipy package out of this? I am happy to help with this. For this, we, however, first have to decide on a folder structure that is more compatible with the pipy framework (see this post)[https://stackoverflow.com/questions/193161/what-is-the-best-project-structure-for-a-python-application]. My suggestion would be the following:

[image]

- **cfg:** contains the configuration `.xml` files.
- **docs:** Contains the sphinx documentation (I am happy to generate this out of the readme [see this example](https://github.com/rickstaa/panda_autograsp)).
- **piano-led-visualizer:** Contains the piano-led-visualizer module (All the classes used in the `vizualize.py` script).
- **resources:** Contains the `.stl` file.
- **scripts:** Contains the `visualize.py` run code.
- **songs:** Contains the midi songs
- **src:** contains the C scripts
- **tests:** contains the python test scripts

In this, the `visualize.py` script would be split into a module and a script that users can run. Let me know your thoughts.

onlaj, 2020-06-12:
Hey, It can be done, but what are the advantages of this solution compared to cloning the repository from github?
Instead of `git clone https://github.com/onlaj/Piano-LED-Visualizer` user will have to type `pip install piano-led-visualizer`, what else will change?

rickstaa, 2020-06-12:
@onlaj You're after thing about it not much will change. The main benefit of adding it to the PyPI repository will be that users do not need to clone the repository but can simply install the package using the `pip install piano-led-visualizer` command. The main benefit of changing the project structure into a model/script based structure would be that we can create a [command-line script](https://python-packaging.readthedocs.io/en/latest/command-line-scripts.html) out of the `visualizer.py` script. This would allow the user to evoke the visualizer using `piano-led-visualizer` instead of `sudo -E python vizualizer.py`.  I am however unsure if that would work when the script needs the `sudo -E`.  The other benefit would be that users can install the package and requirements using one simple `pip install .` command instead. This, however, could already be done by only adding the following setup.py file. The main reason I proposed the project structure change is that I think separating the (third party) modules from the main run code would make it easier for people to contribute to the repository.

vzoltan, 2020-06-12:
@rickstaa 

just create a service :

https://github.com/vzoltan/Piano-LED-Visualizer/blob/master/system_files/lib/systemd/system/pianoled.service

and run it like a service. 
so you can always 

sudo service pianoled start/stop also autostart after boot.

rickstaa, 2020-06-12:
@vzoltan You right that would be a more of an improvement compared to a command-line script. Creating a service would make a pipy package redundant. I will use that on my own raspberry pi for now. Maybe we can add something about that in the documentation.

rickstaa, 2020-06-12:
@vzoltan Or just a note as service creation is already explained in [the rpi-midi-complete documentation](https://neuma.studio/rpi-midi-complete.html)

vzoltan, 2020-06-12:
with what ive linked it runs as service. all i know, it works for me.

rickstaa, 2020-06-12:
@vzoltan It runs perfectly thanks for that! I was thinking of adding something like this to the README.md:

#### Running Visualizer

Download or clone this repository into your RPI.

git clone https://github.com/onlaj/Piano-LED-Visualizer

Using PIP install all libraries listed in requirements.txt file Run visualizer.py with command

    sudo -E python visualizer.py

You can auto run Visualizer on RPi boot, just follow this tutorial: How To Autorun A Python Script On Raspberry Pi Boot. Additionally, you can also create a service that is automatically started when the raspberry boots. This is done by using the 

```bash
sudo systemctl daemon-reload
sudo systemctl enable pianoled.service
sudo systemctl start pianoled.service
````

I was thinking of an addition to the README.md li

rickstaa, 2020-06-12:
@vzoltan It runs perfectly thanks for that! I was thinking of adding something like this to the README.md:

#### Running Visualizer

Download or clone this repository into your RPI.

git clone https://github.com/onlaj/Piano-LED-Visualizer

Using PIP install all libraries listed in requirements.txt file Run visualizer.py with command

    sudo -E python visualizer.py

You can auto run Visualizer on RPi boot, just follow this tutorial: How To Autorun A Python Script On Raspberry Pi Boot. Additionally, you can also create a service that is started at boot. The steps on how to do this can be found below.

##### Create a service
A service can be created for the pianoled script by creating a service startup file  `sudo nano /lib/systemd/system/pianoled.service`:

````bash
[Unit]
Description=PianoLED
After=multi-user.target
Conflicts=getty@tty1.service

[Service]
Type=simple
ExecStart=sudo /bin/sh -c 'HOME=/home/pi /usr/bin/python /home/pi/Piano-LED-Visualizer/visualizer.py'
StandardInput=tty-force

[Install]
WantedBy=multi-user.target
```

Follwing this service can be added to boot at startup using the following commands:

```bash
sudo systemctl daemon-reload
sudo systemctl enable pianoled.service
sudo systemctl start pianoled.service
````

closed 2020-06-12 by rickstaa

rickstaa, 2020-06-13:
@vzoltan I tested your solution out today, and it fully satisfies my need. Thanks again! @onlaj If you need help with any other features just let me know by adding the `help wanted` label. If I find something myself that I think is missing in the future, I will open up an issue.