onlaj/Piano-LED-Visualizer #131: Input Output Ports

harryG15, 2020-06-23:
Hi Onlaj, just wondering - I successfully completed your Build around 6 or 7 months ago but left it for a while... I picked it up again recently and rewired some stuff to make it neater & thought... I should get the new updates from the ‘master’! 

I noticed you don’t have an ‘output’ anymore and my LED strip isn’t working? 
Could either my wiring or maybe I need the output - for it to work?
Let me know
Many thanks

onlaj, 2020-06-23:
Hey.
No, output port is not needed, it was a placeholder. It must be your wiring.

harryG15, 2020-06-23:
Ahh that’s annoying! 
Can I ask?
[image]
Is this wiring ok?

onlaj, 2020-06-23:
Yep, it's good.

harryG15, 2020-06-23:
Sweet! Can I let you know if I get in any more trouble?

onlaj, 2020-06-23:
Sure :)

c4charge, 2020-06-23:
Quick heads up, the raspberry in the video is not a pi Zero. so the LCD connector is "north" instead of "south" if you understand what I mean. (just throwing it out there because I made that mistake lol)

harryG15, 2020-06-23:
[image]
So is this right?

onlaj, 2020-06-23:
Seems to be right, does it work?

harryG15, 2020-06-23:
No, but somethings just occurred to me that I’m powering the Pi & the LED power supply through extension cable!

harryG15, 2020-06-23:
[image]
If you wouldn’t mind taking a look 
Much appreciated

harryG15, 2020-06-23:
Excuse the silly coloured wiring - tried to explain it

harryG15, 2020-06-23:
[image]
This is also direction of arrows?

onlaj, 2020-06-23:
Everything looks fine, it should work :)

closed 2020-06-23 by harryG15