onlaj/Piano-LED-Visualizer #132: help configuring Sevillasoft MIDI USB-USB

stepir, 2020-06-26:
Hi @onlaj 

the MIDI USB interface just arrived and I'm struggling in get it working :-)

The PI is using a OTG cable (4 USB ports): 1 is connected to my USB-MIDI adapter (going to my piano), 1 is connected to Sevillasoft MIDI USB-USB which is then connected to my iPad using the camera adapter (which works if directly connected to my USB-MIDI adapter)

Synthesia see correctly the 4 midi ports (as both input and output devices) exported by Sevillasoft MIDI USB-USB but non of these is correctly working (no input is detected - no output is produced).

Piano-led-visualizer show the 4 input midi ports - what should be used as a config?

The Sevillasoft MIDI USB-USB have the red led blinking but the green led (output I think) never turn on.

Thanks for your help

onlaj, 2020-06-26:
You should use port called "MIDI USB-USB", not the "MIDIIN" or "MIDIOUT".
It seems you connected it correctly, do you have other devices, preferably with Windows 10? I don't have Apple products so I never tested if it works with iPads.

stepir, 2020-06-26:
> You should use port called "MIDI USB-USB", not the "MIDIIN" or "MIDIOUT".

I have 4 MIDI USB-USB, guess just the first one? you mean for the input port of also for the playback one?

> It seems you connected it correctly, do you have other devices, preferably with Windows 10? I don't have Apple products so I never tested if it works with iPads.

unfortunately I have only MAC/IPAD :-(

stepir, 2020-06-26:
I've recabled and now input/output works! perhaps the MIDI USB-USB has specific port as input (PI) and output (iPad)?

The problem I have now is that key lights are now working - are they supposed to right?

onlaj, 2020-06-26:
>The problem I have now is that key lights are now working - are they supposed to right?
Yes, they should. Did you change settings under output port?

stepir, 2020-06-26:
yes! what should be used?

onlaj, 2020-06-26:
"Key lights" should be enabled and set to "finger-based" or something like that.

stepir, 2020-06-26:
yes in synthesia this is set in that way - how about the playback port in piano-led-visualizer?

onlaj, 2020-06-26:
Wait, maybe I misunderstood you. You want to light leds when pressing keys, right? Then you should set inport in Visualizer to your piano.

stepir, 2020-06-26:
I'd like led to be turned on depending on next note (learning mode?) - if I set skipped note to finger based the led turn on (white) when I press the relative key

onlaj, 2020-06-26:
Then in Visualizer settings you should set "MIDI USB-USB" device and in Synthesia "key lights" should be set to finger based.

stepir, 2020-06-26:
I see - it 'works' but unfortunately has exactly the same issue I have with the BT setup.

- notes are badly sustained (i noticed it doesn't happen if I use the piano as input)
- only some notes light the led, 50% (roughly) of the led/notes are not turned on
- the behavior change depending on the song - some have the led working but the notes are horribly cut (the sound is produced for very short fraction of time), some have a better sustain of the note but the led turn on only sometime 

BTW I realized what was the issue about the input not being recognized - if I connect my IPAD after piano-led-visualizer is started it work, if the iPad is already connected it won't.

closed 2020-12-23 by onlaj