onlaj/Piano-LED-Visualizer #134: Feature Request: High level LED color control requests

Kliklijoker, 2020-07-06:
Hi, Aleksander Evening, in following two videos have some special led control technique that I really curious, I think that might be very hard, can you share me with your possible ways to do the following two video effects?

https://www.youtube.com/watch?v=g7C2it9cCsY

https://www.youtube.com/watch?v=lFznCt5kvXM

onlaj, 2020-07-25:
Hey.
The setup from second video is already possible in Visualizer. Just prepare sequence and set one of the pedals as the way to switch between steps.
As for the Rousseau's video, this one is more complicated. I assume he preprogrammed whole song manually.

Kliklijoker, 2020-07-25:
Can you explain more details about "programmed whole song manually"?

How can I change the code to achieve that effect, do I need pre-enter whole song midi into the program?

onlaj, 2020-07-25:
This is just a wild guess, but he could have just prepared in advance an order for colors, like first note is white, second is red, then white etc. Then all he had to do was play it without any mistakes.

Kliklijoker, 2020-07-26:
It sound like a great guess, so can you write an example code about let certain note(like c4) light red then white, and back to red...

Kliklijoker, 2020-08-04:
Hi, Aleksander Evening, in the second video the right hand region and left hand region are vary very often, it very hard to use sequence to continues change the different color light region by pedal, so is it possible have an easier way?

https://youtu.be/b8HVQtIoBYU

Like 3：10~3：20

Kliklijoker, 2020-08-04:
I also find the following video, he mark the mute note as black note.
I thought he just set as right hand led first press light then second dark......, how can I modify the code to get this effect? Anyway, very thank for your crazy project! It's so amazing!

https://youtu.be/34z7xdaaEmk

closed 2020-12-23 by onlaj