onlaj/Piano-LED-Visualizer #135: Bluetooth connect PC problem

Kliklijoker, 2020-07-06:
after issue aconnect -l I got,

client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 20: 'Roland Digital Piano' [type=kernel,card=1]
    0 'Roland Digital Piano MIDI 1'

and I can't see RPi Bluetooth on my PC

onlaj, 2020-07-25:
Hey, try to use Blueman app to connect your PC with RPi.

Kliklijoker, 2020-08-04:
After I try use blueman app I found my RPi zero can't detect any Bluetooth signals, I already try two pc and my phone(Android), in blueman app I can't see any device, is my RPi Bluetooth broken?

Kliklijoker, 2020-08-13:
I already bought a new RPi zero, I very sure is my hardware already broken.

closed 2020-08-13 by Kliklijoker