onlaj/Piano-LED-Visualizer #136: WaveShare LCD only showing white

TheWhitePianoKey, 2020-07-06:
I bought the "Waveshare Raspberry Pi 128x128 Pixels 1.44inch LCD Display Hat with Embedded Controller,Communicating Via SPI Interface"

When connecting it to the raspberry pi, it only shows white. I'm not sure if a driver is missing or something else and how I can fix this. The tutorials I was following just had the screen work immediately without doing any extra steps.

Thanks

onlaj, 2020-07-06:
Did you start visualizer.py script?

TheWhitePianoKey, 2020-07-06:
> Did you start visualizer.py script?
edited my post. It's white even when everything is turned on and the visualizer works perfectly without any delay. Only having problems with the display showing white.

onlaj, 2020-07-06:
Did you enable SPI?

TheWhitePianoKey, 2020-07-06:
I enabled SPI en SSH

TheWhitePianoKey, 2020-07-06:
extra info:
When I connect the pi to the piano and turn it on, it works perfectly.
If I then connect the hdmi (without powering off the pi) I don't get any screen, only black. I have to restart the Pi to make the HDMI to show me anything. So not sure if maybe the HDMI not showing anything is linked to the LCD not receiving any input?

I re-enabled the SPI again using the sudo raspi-config command, just to be sure.

when running the visualizer via manually, the screen still doesn't work.
I started it with: sudo -E python /home/pi/Piano-LED-Visualizer/visualizer.py

TheWhitePianoKey, 2020-07-09:
FIXED: Didn't press the LCD on the pins hard enough. I didn't get the correct screws for it in the package. Ordered a new package which did have the screws and realized the screen needs to be deeper in the pins.

closed 2020-07-09 by TheWhitePianoKey

julien-blanchon, 2020-10-08:
Same issue, I've try to pins harder but still don't work.