onlaj/Piano-LED-Visualizer #146: When I want to change numerical settings (like rgb values) the value automatically decreases

MrBromaba, 2020-08-23:
Hey, basically the title. I can't increase values because the value immediately goes down again. Any ideas why this happens? 
[image]

onlaj, 2020-08-25:
Hey, it looks like your joystick is faulty. Does it also happen in other submenus where you can change settings using left-right clicks?

MrBromaba, 2020-08-25:
Yeah it also happens in other submenus. Do you know whether there's any way to check if the joystick actually is damaged and whether it's reparable? I also noticed that when I run the visualizer the LEDs have quite a bit of a latency and are lagging, especially when multiple notes are played very quickly. This wasn't the case earlier on, so I just had the idea that maybe the faulty stick is constantly sending signals to the pi which causes the LED reactions to be rather slow. Now this is just a wild guess, you are the expert on the technical side, so what do you think? When I'm running the lightkeys.py file in the test folder there is no lag whatsoever. 

Greetings!

onlaj, 2020-08-25:
You can try to disconnect led strip wires, maybe it's them causing the short circuit. If it doesn't help try to reconnect the screen.
Beside that there isn't much you can do, it doesn't look like something you can take apart and fix by yourself. 
If you bought it on Amazon you can try to use your warranty. On Aliexpress it might be more difficult because you will probably have to send it back to China.

MrBromaba, 2020-08-28:
I just reconnected the screen and now it works perfectly. No idea why it didn't earlier on but well, at least now it does :D

closed 2020-08-28 by MrBromaba