onlaj/Piano-LED-Visualizer #153: Suggestion for auto start on boot

Roloo38, 2020-10-11:


onlaj, 2020-10-11:
https://www.raspberrypi-spy.co.uk/2015/02/how-to-autorun-a-python-script-on-raspberry-pi-boot/

Roloo38, 2020-10-11:
Sorry, i forgot to add the coment:)
I saw that link and i think for a begginer(like me) would be much easier to do this:

Hello. If u wanna add a feature for the program to start at the boot of the raspberry u need to write the following command: 

`sudo nano /etc/profile`

And write the following command at the bottom of the file:

`sudo -E python /home/pi/Desktop/Piano-LED-Visualizer/visualizer.py`

My folder with the visualizer script provided by @onlaj is in /Desktop/Piano-LED-Visualzier/
If your folder is in another location just replace the location in command u add at the /etc/profile

It s only a suggestion, i dont wanna make u feel bad or something:)

Let me know what u think about this. Have a nice Saturday!

closed 2021-08-16 by onlaj