onlaj/Piano-LED-Visualizer #154: Flickering LED Animations - Kinda solved

HangryPanda, 2020-10-23:
I was playing with the LED animations and had a really bad problem with flickering with all of the settings and I was perturbed why it was so bad. It would stop flickering when I loaded the pi with actions and the processing slowed. I messed around with some of the settings and I realized that you have to set the LED settings to 'Normal' and not 'Velocity' or 'fading'. 

Also sidenote: the LEDs don't light up on playback on some midi tracks and you have to change the LED settings to 'Velocity" or  "fading' for it to work. 

I can't code, but I can troubleshoot the hell outta something - copy and paste and Googling are my superpowers, lol. I wish I knew python well enough to create a patch, but I digress - if someone can fix these two things that would be awesome! 

User Story 1 : As a user, I would like to enable an LED Animation so that it doesn't flicker without having to change a setting. 
User Story 2: I would like midi playback to light up the LEDs without needing to change the LED setting from Normal to work. 

Thanks all!

onlaj, 2020-10-24:
>I was playing with the LED animations and had a really bad problem with flickering

I didn't notice anything like that, can you record it and post here? Which RPi do you use?

> Also sidenote: the LEDs don't light up on playback on some midi tracks and you have to change the LED settings to 'Velocity" or "fading' for it to work.

Can you show me example midi files? I just tested some of my own and they work fine with fading/velocity.

HangryPanda, 2020-11-03:
It was the capacitor causing the flickering. But I also noticed it on the RPi 4 when powering it using the USB C power port. You definitely want to supply power to the RPi using the GPIO pin. I have no issues with flickering there. 

As for more comment about slow playback on RPi4 - you need to install Python3 and install all the dependencies in the requirements.txt file using pip3. After that everything works great. I haven't tested Bluetooth because I've never been able to get Bluetooth to work well on both the RPi 4 and Zero. Too much latency and many skipped notes. I use the Sevilla cable and it works wonders. I don't know how well it works on ipad since I haven't tested it.

closed 2020-11-03 by HangryPanda