onlaj/Piano-LED-Visualizer #157: LEDs stop reacting while changing inport to piano

Hicershice, 2020-10-27:
Hi !
i did not use my LEDs for like 3 months, now i wanted to use it again and when i start the program on my rpi, i have have to change the input to my piano as always. When i did this today, i faced the issue, that changing the inport to my piano it took a few seconds longer than usual and while its changing i can see the LEDs lighting up, but when it finishes changing the LEDs stuck and dont react anymore. 
do you know what the issue can be ?

greetings

Hicershice, 2020-11-06:
I now got a closer look to the problem.
the cable and the midi output from the piano are working fine.
the leds and the raspberry are working fine too.
If i tell the rpi to do some LED animations it looks great, but as soon as i want to play directly or play MIDI from the rpi, no led is lighting up.
Please help me, i cant figure the problem.

Hicershice, 2020-11-11:
I had to solder the cables again to the rpi , the Connection was Bad. Now it doesn't flicker anymore . But now the problem is a bit different:
After 10-300 seconds the LEDs freeze. When I set the import to my piano again the LEDs start to work again for like 10-300 seconds...
I don't understand it ! How can this be possible. Why do they stop working and when I set the import to my piano (again!) The work again. .is it changing the import automatically?

PianoBear3, 2020-11-12:
I have a similar problem

onlaj, 2020-11-12:
Which version of Visualizer do you use? Is it the latest one? Did it work on older versions?

@PianoBear3 can you close your issue? Let's try to fix it in this thread :)

PianoBear3, 2020-11-12:
@onlaj  sure :) i use the latest version, don't know about the older versions :/

Hicershice, 2020-11-12:
@PianoBear3 at least I'm not alone :D

@onlaj I have the latest version installed 4 days ago. The older version had the bug I posted, but it was because of soldering I think.. while fixing the soldered cables I also installed the new version.

2 weeks ago it was working with the version I installed 1 year ago

onlaj, 2020-11-12:
Can you try older version?
`git checkout 84f53c945f54d8f03e464231cf7865d817960cb4`

This is version from over one year ago.

PianoBear3, 2020-11-12:
I dont understand how Tod User gibt checkout..

When i use it, It just says „HEAD is Now at 84f53c9 Add backlight Feature“

PianoBear3, 2020-11-12:
Wow, it works fine for me!
But why not the new version? Its even cooler..

onlaj, 2020-11-12:
That's what I'm trying to figure out :) 
If you have time, can you try newer commits? I want to know which one caused the problem. You don't have to try it one by one, just skip few of them until this error appears.
I can't reproduce this error on my side so I can't test it properly.

PianoBear3, 2020-11-12:
Sure! Just Tell me what i have to type ;)

onlaj, 2020-11-12:
Try
`git checkout 6dbe2cf64bb7f9b5e6ba77a7789da89d3487785a`

PianoBear3, 2020-11-12:
Works fine

onlaj, 2020-11-12:
`git checkout 89619a5a4ba8eccb9654222d9bfd68a8a237081b`

PianoBear3, 2020-11-12:
Seems to work too

onlaj, 2020-11-12:
Try this one
`git checkout 43db2a8adf974b53c895ac4d973ba4e5bce6b55f`
and then you can go back to the latest one.

PianoBear3, 2020-11-12:
The Version u Worte and the latest Version are working Fine .. i dont know how, but the problem seems to be solved ^^

Thanks a lot for your fast answers and helping!

Hicershice, 2020-11-12:
Works for me too ... Just needed to be reinstalled I guess 😁

onlaj, 2020-11-12:
Well, I shoud've thought about recommending that to you in the first place, I'm glad it works now :D

Hicershice, 2020-11-22:
Well now I have the same problem again.
I start the program and all works fine. After 1 minute the LEDs stop lighting up and when I change the input back to my piano, they work for 1 minute and stop again .

onlaj, 2020-11-23:
Try to reinstall again.
What changes do you make inside the visualizer settings? I'm trying to figure out what could be causing this problem.

Hicershice, 2020-11-23:
I reinstalled it 6 times now .. does not seem to help .
Sometimes I don't change anything , it still stops after random amount of time .
Looks like it stops or changes the input by itself because I can switch it back on in input settings but only for a few seconds or minutes again.

I checked the midi cable from piano to rpi, it's all good and the piano midi output is okay as well..

onlaj, 2020-11-23:
Last time reinstalling helped both of you. How do you do it, delete all files and then git clone them again? Might be a dump question to ask but did you try to restart RPi?

Hicershice, 2020-11-23:
Yeah I delete them all and clone again.
I also rebooted the rpi after the clone and tried git checkout with some branches but it's always the same.
Maybe it's the rpi itself?

Hicershice, 2020-11-24:
Figured the problem out.
My piano can't have an output of 2 midi connections. If so , the rpi closes the input after a time cause the other midi recorder is active, that's why I can switch it back on.

Thanks for your help :)

closed 2020-11-24 by Hicershice