onlaj/Piano-LED-Visualizer #158: Numpy doesn't work.

HangryPanda, 2020-10-28:
`  File "/home/pi/Piano-LED-Visualizer-JV/visualizer.py", line 5, in <module>
    import LCD_1in44
  File "/home/pi/Piano-LED-Visualizer-JV/LCD_1in44.py", line 31, in <module>
    import numpy as np
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/__init__.py", line 145, in <module>
    from . import lib
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/lib/__init__.py", line 9, in <module>
    from .index_tricks import *
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/lib/index_tricks.py", line 13, in <module>
    import numpy.matrixlib as matrixlib
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/matrixlib/__init__.py", line 6, in <module>
    from .defmatrix import *
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/matrixlib/defmatrix.py", line 13, in <module>
    from numpy.linalg import matrix_power
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/linalg/__init__.py", line 51, in <module>
    from .linalg import *
  File "/home/pi/.local/lib/python2.7/site-packages/numpy/linalg/linalg.py", line 34, in <module>
    from numpy.lib.twodim_base import triu, eye
ImportError: No module named twodim_base`

Any time my RPi Zero tries to start the screensaver I get this error. Not sure how to fix it. 

Thanks!

onlaj, 2020-10-29:
Try to reinstall numpy.

HangryPanda, 2020-11-03:
that worked! Thanks!

closed 2020-11-03 by HangryPanda