onlaj/Piano-LED-Visualizer #159: Visualizer could be more responsive

ak48disk, 2020-11-01:
I have made my visualizer using this code , thanks! I discovered that the visualizer sometimes have delays, and staccato was not shown. I have changed the code, namely two places, to make it more responsive:

- in neopixel.py, only call ws.ws2811_render() when data is changed ([example](https://github.com/ak48disk/Piano-LED-Visualizer/blob/master/neopixel.py))
- in the main loop, do not render the menu or lcd, if keys are recently pressed. Only render it after the piano has been idle for more than 2 seconds. This ensures that keypresses are always processed first.

I have also changed the code in my fork so that the visualizer accepts midi over TCP, so I can connect my piano to the Windows tablet and run a small client there to forward midi to my pi. This is useful for me because I am using a virtual instrument (PianoTeq) on my tablet to render the sound, not in  the piano itself. However it was not done cleanly and I have ignored all the settings, etc, so that a merge here needs much more work.

onlaj, 2020-11-01:
Hello.

> in the main loop, do not render the menu or lcd, if piano are recently pressed

It's already done [L1969](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L1969)
Besides, it only refresh every 60 loop cycles (which is too high IMO, scrolling long MIDI file titles takes too long, I might change that), so I don't think it cause any delays or stutters.

>in neopixel.py, only call ws.ws2811_render() when data is changed

I will take a look into that, thanks.

> so I can connect my piano to the Windows tablet and run a small client there to forward midi to my pi

Did you make your own client? I tried RTPMidi but it didn't work.

ak48disk, 2020-11-01:
I used sample code here from mido: https://mido.readthedocs.io/en/latest/socket_ports.html, loop in PortServer didn't work, I need to accept() and call client.iter_pending() like the code at the bottom of the page.

also if connect through wifi, need to disable wifi power management for better performance, I think a wired ethernet connection will be much better but the zerow don't come with an Ethernet port

sudo iwconfig wlan0 power off

closed 2021-08-16 by onlaj