onlaj/Piano-LED-Visualizer #162: rtpMIDI configuration how?

xxx02, 2020-11-18:
I managed to connect raspberry pi to windows pc using rtpMIDI with the help of this guide: https://blog.tarn-vedra.de/pimidi-box/ The latency and connection quality is just perfect. But now I'm in trouble with set ports (MIDI IN&MIDI OUT) on raspberry pi. The piano contacts with pc over raspberry pi using wireless connection. It's ok. But on pi I can't select my piano as an input. It gives error. It's possible to select it as an playback port. I just don't know what's going on background. I'm newbie with coding by the way. Just followed the instructions and I got here 😊

onlaj, 2020-12-01:
Create virtual midi ports using 
`modprobe snd-virmidi `
and then use `aconnect` to make a bridge between your piano midi input/output and rtp connection with one of the virtual ports in between.
I know it's far from perfect, use it as temporary solution, I'm working on implementing rtpmidi into my script, no ETA yet.

xxx02, 2020-12-02:
That's fantastic news! Thank you very much 🙏

closed 2020-12-02 by xxx02