onlaj/Piano-LED-Visualizer #164: LCD white screen

toast122, 2020-11-19:
Hi, when i power up my raspberry pi 4, all it shows is a white screen. I am using the 1.44in LCD display

xxx02, 2020-11-19:
Try this steps:
Configure Visualizer to run at boot
"sudo nano /etc/profile"
Paste in the following at the bottom of the file:
"sudo -E python /home/pi/Piano-LED-Visualizer/visualizer.py &"

toast122, 2020-11-20:
I have tried that, but the issue still persists

any other reccomendations? 

anything helps :) thank you

toast122, 2020-11-20:
Hi, i was able to see what the problem was. i pasted sudo -E python /home/pi/Piano-LED-Visualizer/visualizer.py & in the wrong location near the bottom of the file. It works perfectly fine now

Thank you for your help :)
I appreciate it a lot

closed 2020-11-20 by toast122

ChoPraTs, 2025-03-03:
Hello. I have the same problem, could anyone help me, please?

I have never used a Raspberry until today and I am a little lost.

I've bought all the components mentioned in the tutorial and I've already managed to install the software through Etcher and the System Image provided.

Thanks to one of my piano students, everything is already connected, the LEDs are working and receiving the MIDI data from my piano correctly. I can enter to the software through the Wifi Hotspot but the LCD screen stays blank.

How should I put those commands and where exactly? Do I need another software or to connect the Raspberry to my computer? I'm macOS user.

Thank you so much.

ChoPraTs, 2025-03-04:
Maybe there's a problem with version 1.6 of the software and the screen. I have tried by installing the previous version 1.5.1 (from 2023) and it seems that it has solved the problem, now the LCD screen works without problems.

The next problem I have is with LED lights, I think they are not functioning as it should, as I could see in several videos. In my case, the colors are not corresponding to those that I select and I can perceive a strange flickering of the LEDs adjacent to the correspondent to the key that I press that I think should not happen either. I will open another independent thread to try to deal with this issue, because I bought two different LED strips and is happening the same on both.