onlaj/Piano-LED-Visualizer #165: Freezes in "System Info"

xxx02, 2020-11-26:
When I click the "System Info" menu it freezes and give this output:
```
pi@raspberrypi:~/Piano-LED-Visualizer $ sudo -E python visualizer.py
Inport set to ARIUS:ARIUS MIDI 1 20:0
playport set to ARIUS:ARIUS MIDI 1 20:0
Traceback (most recent call last):
  File "visualizer.py", line 2000, in <module>
    menu.enter_menu()
  File "visualizer.py", line 851, in enter_menu
    menu.change_settings(self.current_choice, self.currentlocation)
  File "visualizer.py", line 1089, in change_settings
    screensaver()
  File "visualizer.py", line 1417, in screensaver
    menu.render_screensaver(hour, date, cpu_usage, round(cpu_average,1), ram_usage, temp, cpu_chart, upload, download, card_space)
  File "visualizer.py", line 899, in render_screensaver
    fonthour = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeSansBold.ttf', self.scale(31))
  File "/home/pi/.local/lib/python2.7/site-packages/PIL/ImageFont.py", line 640, in truetype
    return freetype(font)
  File "/home/pi/.local/lib/python2.7/site-packages/PIL/ImageFont.py", line 637, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home/pi/.local/lib/python2.7/site-packages/PIL/ImageFont.py", line 186, in __init__
    font, size, index, encoding, layout_engine=layout_engine
IOError: cannot open resource
pi@raspberrypi:~/Piano-LED-Visualizer $
```

onlaj, 2020-11-26:
Install font
`sudo apt-get install fonts-freefont-ttf`

xxx02, 2020-11-26:
Voila! Thank you 🙏

closed 2020-11-26 by xxx02