onlaj/Piano-LED-Visualizer #167: Better way to start the script at boot

david4599, 2020-12-06:
Hi, I think the script shoudn't be started from /etc/profile because each time you log into the raspberry pi, it will be launched.
And this may be a problem because if we connect to the pi with PuTTY, the script will be launched a second time and even more if we connect to multiple sessions. You can see in the screenshot that I have 3 instances of the script running for each opened session.

[image]

Of course, these unwanted instances are killed once the PuTTY sessions have ended so this is not a huge issue but it's a little bit annoying because just after I loged as root, I got a message "No module named 'spidev'" IIRC. And after a moment, I figured out the script was started again and the requirements was not installed for the root user that's why this message showed up.

To solve this problem, I created a service to run the script like you did for the midi service in the manual installation. I just had to install the script's requirements again as root this time and set the ExecStart command like this: `ExecStart=python3 /home/Piano-LED-Visualizer/visualizer.py`
\
\
\
Just before send this message, I tested the script running multiple times again and there is definitely a problem that I just realized. The midi messages are received by all the instances of the script and the LEDs are controled by each instance so there is a general lag of the LEDs while playing on the piano.
Also, maybe an easier solution I just thought is to keep the start command in the /etc/profile but modify the script to end if another instance of the same script is already running.

onlaj, 2020-12-06:
> modify the script to end if another instance of the same script is already running

I thought [this part](https://github.com/onlaj/Piano-LED-Visualizer/blame/5eb64a414e7a93b42e68bc296378c577146f93b2/visualizer.py#L28) was responsible for it, no? I didn't write that code but it seemed to work for me. However, I never log in to the root account, always use 'pi' user.

david4599, 2020-12-06:
Yeah, but I think this part doesn't work properly, at least for me. I've added 2 debug print and it seems to always restart the script in loop the second time I launch it.

[image]

[image]

The issue here is that the script shouldn't be restarted because if we can't lock the file, it means the script is already opened an running. So why restart it?

Maybe the simple fix is to replace `restart_script()` by `exit()` and delete the restart_script() function? Or am I wrong?

Also, launching the script as a service doesn't allow multiple instances because I think systemctl prevent that, I've just tested it by manually start the service many times and there is only one instance running so this can be another fix.

onlaj, 2021-09-13:
I changed method of starting script on boot to crontab, it doesn't start the script when you ssh to RPi now.

closed 2021-09-13 by onlaj