onlaj/Piano-LED-Visualizer #168: Led Color Confusion on Synthesia Learning

xxx02, 2020-12-06:
I have successfully connected pi with pc over RTP. Thanks for that @onlaj
I have 2 questions:
- Synthesia blue and green color works but when a note comes the first time it lights up default color not blue or green on the second come of the same note it's ok. Every new notes just light up default color. I have tried every option in skipped notes as mentioned but didn't work at all.
- Is there any way for assigning ports without using terminal. We have to use the command `aconnect  20:0 128:1` and `aconnect 128:1 20:0` every time after restart. Maybe there can be an option in visualizer menu? And also for restarting rtpmidi and resetting port connections.

onlaj, 2020-12-06:
- Does this happen with every midi file?
- I'm working on that

xxx02, 2020-12-06:
I just tried only synthesia's songs.

luizzitol, 2021-02-21:
I am experiencing exactly the same on Syntesia. The first time a note is coming will get default color and then it will go to green or blue... Then in the middle of the song it will randomly use the default color instead of the green/blue from Syntesia.

Did you find any solution to this xxx02?

xxx02, 2021-02-21:
Not yet :/

closed 2021-08-16 by onlaj