onlaj/Piano-LED-Visualizer #170: VISUALIZER MENU not showing on Rpi screen

ketsogo, 2020-12-13:
I used the LED Visualizer System, updated it, connected to wifi and enabled ssh but after rebooting it nothings shows up on the screen even after 30 mins.

onlaj, 2020-12-13:
Can you connect to ssh? Try to launch script manually.

ketsogo, 2020-12-14:
yes I can connect to ssh, I enabled ssh via the config settings... Do you mean manual installation?

devucsb, 2020-12-14:
Have you made any modifications to config.txt or cmdline.txt? How are you trying to SSH?

I was unable to setup ssh via wifi so I researched alternatives on my Pi Zero W I added modules-load=dwc2,g_ether to cmdline.txt, and dtoverlay=dwc2 to config.txt, and that resulted in the same behavior for me.

onlaj, 2020-12-15:
>Do you mean manual installation?

No, just connect to ssh, navigate to script folder and launch script.

`cd /home/Piano-LED-Visualizer`
`sudo python3 visualizer.py`

closed 2021-08-16 by onlaj