onlaj/Piano-LED-Visualizer #171: Fixing PLV screen resolution

ketsogo, 2020-12-14:
The led visualizer system showed up but it just looks like this...and btw how do you transfer recorded midi to pc wirelessly??
[image]

onlaj, 2020-12-15:
Try to reattach screen hat, looks like faulty connection.

>and btw how do you transfer recorded midi to pc wirelessly??

 Via SSH using some ftp client.

ketsogo, 2020-12-15:
Ok thanks... I'll try if it works

ketsogo, 2020-12-22:
I tried many times but it still doesnt work

onlaj, 2020-12-22:
Seems like your hat or RPi might be damaged. Try to download some demo code for this screen and test if it works https://www.waveshare.com/wiki/File:1.44inch-LCD-HAT-All-Code.7z

ketsogo, 2021-01-11:
How do I use this code??.. Sorry just a beginner hehehe

onlaj, 2021-01-11:
@ketsogo download the 7z file, unpack it and transfer RaspberryPi\python\ to your RPi.
Then launch the demo code with
`sudo python key_demo.py`

ketsogo, 2021-01-12:
Do I transfer it on the SD card?

onlaj, 2021-01-12:
This or just use sftp. Put it in /home directory and launch the demo.

ketsogo, 2021-01-13:
The file you sent is for 1.44 inch ... is it okay if my lcd hat is 1.33?

onlaj, 2021-01-13:
Either I missed your mention of the display type or you didn't mention it at all. If you want to run a script on a 1.3 inch screen you need to add the appropriate parameter, did you do that?

> If you are using WaveShare 1.3inch 240x240 LED Hat instead of 1.44inch 128x128, add this instead: sudo python3 /home/Piano-LED-Visualizer/visualizer.py --display 1in3 &

https://github.com/onlaj/Piano-LED-Visualizer/blob/master/manual_installation.md

ketsogo, 2021-01-13:
yes I added the appropriate one for the 1.33

onlaj, 2021-01-15:
In that case get demo code for your display and test it https://www.waveshare.com/wiki/File:1.3inch_LCD_HAT.7z

closed 2021-08-16 by onlaj