onlaj/Piano-LED-Visualizer #174: visualizer menu

darmap, 2020-12-17:
why every time you choose system info on the visualizer menu is always stuck

onlaj, 2020-12-17:
Probably you don't have font installed.
Connect to ssh and type
`sudo apt-get install -y fonts-freefont-ttf`

If it doesn't help navigate to project folder
`cd /home/Piano-LED-Visualizer`
launch script
`sudo python3 visualizer.py`
and then try to open system info. If screen freeze it should output some error in your ssh console. Copy and paste it here.

darmap, 2020-12-19:
after I select system info back to the start menu

darmap, 2020-12-23:
after I select the system info ssh will display like this 
XDG_RUNTIME_DIR (/run/user/1000) is not owned by us (uid 0), but by uid 1000! (This could e g happen if you try to connect to a non-root PulseAudio as a root user, over the native protocol. Don't do that.)

darmap, 2020-12-23:
after I select the system info ssh will display like this 
 Traceback (most recent call last):
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 1995, in <module>
    menu.enter_menu()
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 851, in enter_menu
    menu.change_settings(self.current_choice, self.currentlocation)
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 1087, in change_settings
    screensaver()
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 1381, in screensaver
    temp = find_between(str(psutil.sensors_temperatures()["cpu-thermal"]), "current=", ",")
KeyError: 'cpu-thermal'

onlaj, 2021-01-12:
Sorry for late reply.
Make sure your Piano-LED-Visualizer is up to date.