onlaj/Piano-LED-Visualizer #175: LED lights no response

ketsogo, 2020-12-22:
I followed your wiring instructions but the LED lights doesn't respond when I press a key.... my wires looks like this btw or if you if you could explain were should I connect each wire,, I would appreciate the help
[image]

onlaj, 2020-12-22:
>I followed your wiring instructions

Well, it doesn't look like it :)
Look at this [picture](https://camo.githubusercontent.com/3ec09e798ccd814bdc16c28134f07e96dfaf52b33ef90d61e5617acb0223a613/68747470733a2f2f692e696d6775722e636f6d2f4a3177413172552e6a7067)
or [this](https://tutorials-raspberrypi.de/wp-content/uploads/Raspberry-Pi-WS2812-Steckplatine-600x361.png)

You have 5 wires coming from your led strip, ignore the red and white pair. From the remaining 3 set pair Red one has to go to (+) plug in your socket.  White one goes to both RPi (ground pin) and (-) plug in power supply socket. Green/Blue goes to RPi pin (GPIO 18).

ketsogo, 2020-12-22:
Ohh I tought the red and white pair will be used and I followed the intructions badly hehe :) .... So do I need to connect another wire to the white wire to join them?

ketsogo, 2020-12-23:
And another issue is that the buttons in the LCD hat doesn't respond when I press them

ketsogo, 2021-02-08:
https://user-images.githubusercontent.com/75939484/107238473-d6527700-6a62-11eb-9931-425a34c94392.mp4

I have connected all the wires correctly but the light still doesn't respond.. I am using usb midi in/out btw, could this be the problem?

closed 2021-08-16 by onlaj