onlaj/Piano-LED-Visualizer #178: LEDs not responding

nealson-s, 2020-12-30:
Hello, absolute beginner here.

This is my first time ever using a pi, python, or wiring a circuit such as this. I have successfully installed and loaded visualizer on the pi however I am having trouble getting the LEDs to light up at all (let alone respond to keys). Here are some pictures of my setup. 
[image]
[image]
[image]
(Where the white wires join I twisted and soldered all 3 together)
[image]

It might be worth noting that the midi connection is seemingly good, as I heard La Campanella coming through my piano. 

This is my second time configuring this setup. I soldered the wires to the bottom this time instead of wrapping them around the ground and GPIO 18 pins as well as soldered the 3 white wires together (I thought these could be the problem). 

I honestly don't know what could be the issue here so if anyone more experienced could help me out, I would really appreciate it. I can provide more information or pictures as well if you need them.

onlaj, 2020-12-31:
Hey.
Wiring looks good to me.
Try to run [tests/strandtest.py](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/tests/strandtest.py) to make sure it's not my script's fault. It's not likely but won't hurt to make sure.
If it's not software problem, probably one of those is damaged:
- led strip
- power supply
- rpi (gpio pin/pins)

Do you have multimeter to check each component?

What exactly can I see on the first photo, do you have two led strips connected together?

nealson-s, 2020-12-31:
Thanks for your response. So I can just paste the code from test/strandtest.py into terminal after connecting to the pi via ssh to run it? Sorry for my lack of knowledge on this.

I don't have a multimeter, should I get one to test each potential problem?

Yes you are correct, I connected the two 1m LED strips together because I wanted to be sure everything worked before I solder them together.

Also something interesting happened. I set it up again without knowing that the white wire had disconnected from the pi (probably due to bad soldering or me moving it too much) and 7 of the LEDs lit up when I plugged in the power supply. I then touched the white wire to the ground pin again and the number of lit LEDs doubles or so. Not sure what this means, although I guess we know the power supply and LEDs work. Here is a picture as well.
[image]

onlaj, 2021-01-01:
> So I can just paste the code from test/strandtest.py into terminal after connecting to the pi via ssh to run it? Sorry for my lack of knowledge on this.

Or just navigate to tests folder
`cd /home/Piano-LED-Visualizer/tests`
and then run test 
`sudo python strandtest.py`

We know for sure that PSU is working and LED strip can light, but the question is if the data can be transmitted properly. I'm not sure how you can test it without multimeter tho, quick google search showed me this [script](https://elinux.org/R-Pi_Troubleshooting#Testing), maybe it's worth giving a try.

btw. just to make sure: did you enable [SPI](https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/) and [disable audio](https://github.com/onlaj/Piano-LED-Visualizer/blob/master/manual_installation.md#5-disabling-audio-output)?

nealson-s, 2021-01-02:
Ran the test... got this:
Traceback (most recent call last):
  File "strandtest.py", line 9, in <module>
    from neopixel import *
ImportError: No module named neopixel

I gave the gpiotest script a try as well, except when I tried to download it to the pi, I got denied permission for some reason. Would it be worth buying a cheap multimeter in this case?

And yes, I do remember enabling SPI and disabling audio for sure.

[image]

onlaj, 2021-01-02:
try also
`sudo -E python strandtest.py`
or with python 3
`sudo python3 strandtest.py`
or
`sudo -E python3 strandtest.py`

nealson-s, 2021-01-02:
Tried all three commands and got the same result as above. Just wondering, do I need everything connected up to the pi when I run the strandtest?

Finally got the gpiotest to work and ran it a couple times...gpio 18 failed every time. I guess this is the problem then?

onlaj, 2021-01-02:
>do I need everything connected up to the pi when I run the strandtest?

Yes, but it should at least launch the script without errors, no matter if cables are connected or not.
You can try to follow software part on this [guide](https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/), maybe you need to clone rpi_ws281x repository to launch strandtest.py

>Finally got the gpiotest to work and ran it a couple times...gpio 18 failed every time. I guess this is the problem then?

Did you unsolder the data cable before running the script? If yes and it showed error on pin 18 it's probably damaged. You would need multimeter to be sure.

nealson-s, 2021-01-02:
Okay, I set it all up again and followed the software instructions in the link and got the test running, however it just kept repeating color wipe, theater chase, and rainbow animations for like 20 minutes so I quit the process out of impatience. And while it was running none of the LEDs lit up like in the video he attached. I assume this is because of pin 18?

Yes I unsoldered everything before running the pin test. I sent in for a free replacement through amazon for the pi so I will be getting a non-defective one shortly. Do you think that should solve the issue?

onlaj, 2021-01-12:
> Yes I unsoldered everything before running the pin test. I sent in for a free replacement through amazon for the pi so I will be getting a non-defective one shortly. Do you think that should solve the issue?

It should. If not, it means that the led strip is damaged.

nealson-s, 2021-01-20:
Got my new pi, set everything up and soldered the leds together...works perfectly now. I seriously can't thank you enough for putting this amazing project together and helping me troubleshoot my issues. I am suggesting this project to all my friends. Thanks again man.

closed 2021-01-20 by nealson-s