onlaj/Piano-LED-Visualizer #180: Learning without Synthesia

EmanuelFeru, 2021-01-06:
Hello,
I want try this project and before I buy all needed components, I want to ask you if it is possible to learn without using Synthesia.
I just want to:
- load an existing midi file
- learn with illuminated keys while following a Sheet music
- the illumination should wait until I press all illuminated keys in order to continue

Is this something supported already?

I checked most Closed/Open issues couldn't find the and answer, but maybe I missed it. Looking forward to your reply.

onlaj, 2021-01-15:
Hey.
Maybe I will add this feature in future :)

EmanuelFeru, 2021-01-16:
It would be nice. Thank you. I think it is very useful to not having the need to be tight to Synthesis. 

Currently i managed to have the screen with the visualizer running. I am waiting for the LEDs to arrive.
Meanwhile I have a tried to run the .mid song connected to the USB port of my piano. Unfortunately, i didn't hear the song playing.. not sure why. Do I need to do some configurations for the midi port in Raspberry pi side?

EmanuelFeru, 2021-01-17:
Hi again @onlaj,
From high level point of view, what are  the steps to implement Learning without Synthesia? I would like to give a try myself into this.

EmanuelFeru, 2021-01-17:
**Short update**

Positive:
The song `La Campanella.mid` is playing on my piano Casio Privia PX-360M... and that is great! As was written in another thread, the piano needs to be started before the Raspberry Pi 0 (RPi0) boots.

Some remarks:
* If no MIDI port is detected at RPi0 boot, then periodically refresh ports and connect if a MIDI port becomes available. This will solve the issue of having the Piano started before RPi0.
* The song  `La Campanella.mid` in my case takes exactly `35 seconds` to start after I press Play in RPi0. Any idea why is it so long?

onlaj, 2021-01-20:
> From high level point of view, what are the steps to implement Learning without Synthesia? I would like to give a try myself into this.

You need to implement your own way of reading midi files. Now I'm using play method from mido library which loads whole song into RAM before playing (it explains why you have to wait after pressing play). If you want to light next-to-play key you need new method which will hold play until the right key(s) is/are pressed.

> the piano needs to be started before the Raspberry Pi 0 (RPi0) boots.
Did you try to replug the usb cable?

EmanuelFeru, 2021-01-20:
Thank you for the reply. I will check the mido library to see how could we achieve sequential playing.

I did not try to replug the wire :)

EmanuelFeru, 2021-02-02:
Hi @onlaj I was looking at the code below which does the time delay to Play the MidiFile:
https://github.com/onlaj/Piano-LED-Visualizer/blob/06473d2513b723212953afba0506a7a52c82277b/visualizer.py#L1274-L1286

1. I am just wondering why is the `delay_debt`  and offset `- float(0.003)` needed here? And just using `time.sleep(delay)` is not sufficient. Maybe, I am missing something.
2. And another thing is how come that while playing the song I am still able to browse through the LCD Display menu? I thought using `time.sleep()` is blocking. Where is the trick I am missing?

onlaj, 2021-02-02:
@EmanuelFeru 
1. When two notes are played at the same time delay between them is equal 0, but it still takes time to play them. Even if it's 1 ms this time is accumulating, that's why on next note (if it's possible) the delay is reduced. As for 0.003, my testing showed that this was needed to get more accurate playback times.
2. threading

EmanuelFeru, 2021-02-02:
Makes sense. I have just seen indeed that you used Threading. I will do the similar, the only difference will be that I need to interact with the Thread. Basically:
- I need to send the currently pressed list of notes from the `while True:` loop to the Thread. Do you know how to achieve this?

EmanuelFeru, 2021-02-06:
@onlaj How do you develop for the LED visualizer on Raspberry pi? What is the process you take?

EmanuelFeru, 2021-03-09:
Hi @onlaj,

Learning without Synthesia is implemented. Check the PR above. Overall it came out very nice. Have a look and tell me what you think.

Here is a picture of the menu:
[image]

closed 2021-03-11 by EmanuelFeru