onlaj/Piano-LED-Visualizer #184: Is persistent piano to rpi connection possible?

msaeedm, 2021-01-20:
Hi @onlaj 

Thank you for this amazing project. 

I recently got the Sevilla's Soft MIDI USB-USB device to connect my iPad. This worked well.  I found it tempting to be able to use persistent connection for my piano..and by persistent I mean no need to refresh ports every time I turn on and off the piano. 

**_Is it possible to connect both the piano and the iPad to the USB-USB device which in turn is connected to the rpi?_** I tried connecting the piano (usb to host) to it but could not get it to work. I tried the all 4 ports added by the device with no luck, There is also no entry like 'portable grand' that is usually added when connecting without the usb-usb device.

Any suggestions?

onlaj, 2021-01-20:
> Is it possible to connect both the piano and the iPad to the USB-USB device which in turn is connected to the rpi?

It's not only possible but this is the way how you suppose to connect everything.

My setup looks like this: 
Piano (via MIDI-USB interface) <-> USB OTG HUB <-> RPi
Tablet <-> Sevilla's Soft USB-USB <-> USB OTG HUB <-> RPi

Then in Visualizer menu I choose my piano (in my case it's mio midi-to-usb interface) when I want to light leds on piano press or first "MIDI USB-USB" when I want to light next-to-play keys.
As long as Sevilla's Soft device is powered (I think you need to connect RPi to left port on USB-USB device) Visualizer remembers the inputs and I don't have to refresh ports.

msaeedm, 2021-01-20:
> My setup looks like this:
Piano (via MIDI-USB interface) <-> USB OTG HUB <-> RPi
Tablet <-> Sevilla's Soft USB-USB <-> USB OTG HUB <-> RPi

This is pretty much my setup too but Visualizer does not remember the piano connection. I have to refresh every time the piano is turned on. ??!! What am I doing wrong?

onlaj, 2021-01-20:
> The output label at the back of the piano is 'usb to host'. Is that the same as midi-usb interface?

Yep.
I assume it happens because when you power off your piano the "usb to host" is also disabling. In my setup RPi powers MIDI-USB interface so it keeps connection.
I will try to make some changes in code, so it will attempt autoreconnect when losing connection.

msaeedm, 2021-01-20:
thank you sir

msaeedm, 2021-03-29:
@onlaj will you pretty please look into fixing this bug soon

onlaj, 2021-03-29:
@msaeedm No matter what I do I can't reproduce this problem on my side, even when I disconnect the cable it reconnects on itself when I reattach usb. 
I made a branch, can you fetch it and test if it works for you? The reconnect feature is activated when you exit screensaver. You can manually start screensaver under Other Settings -> System Info
https://github.com/onlaj/Piano-LED-Visualizer/tree/reconnect
If you don't know how to fetch branch here is instruction: https://www.freecodecamp.org/news/git-checkout-remote-branch-tutorial/

msaeedm, 2021-03-29:
I fetched and tested the new branch...I made a small edit to the code to make it work. I will make a PR to better explain it.

Thank you for your quick response

onlaj, 2021-03-30:
I see, but does it fix problem with your piano?

msaeedm, 2021-03-30:
@onlaj Yes that fixed the problem. Thank you. Now I just exit the screensaver and the lights will work again.

I am wondering if it would be useful for my situation and possible others having the same issue if a function in the visualizer script can be made to evaluate the health of the connection and reconnects if needed. The reconnect method you made forces a reconnect on every screensaver exit. I understand this will be difficult without you even being able to reproduce the problem in your setup. For what it is worth, I will experiment debugging the code to hopefully narrow down where the issue might be. The only limitation right now with this forced reconnect is that I can not exit the screensaver by a piano key press which is a feature introduced in a recent commit.

onlaj, 2021-03-31:
> The only limitation right now with this forced reconnect is that I can not exit the screensaver by a piano key press which is a feature introduced in a recent commit.

I guess it didn't work earlier too, right? You couldn't exit screensaver by piano press if you restart your piano in the meantime.
If we can find a way to detect the lost connection, we can create handler to detect it and force reconnect_ports once new (old but reconnected) port is available. I don't want to trigger that function too often especially if it's not a common problem.

msaeedm, 2021-04-01:
Yes that is correct. I could not exit the screensaver by piano press. I wonder how come my set up is doing this and how come it is not reproducible or similarly reported by others

closed 2021-05-15 by onlaj